prusa3d/Prusa-Firmware-Buddy #3847: [BUG] G90 not working as for MK3 in MK3-compatibility mode

thomasloven, 2024-03-21:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3.9

**Printer firmware version** - 5.1.2+13478

**Original or Custom firmware** - Original

**Optional upgrades** - E3d Revo heater and nozzle

**USB drive or USB/Octoprint**
  Flash drive

**Describe the bug**
When trying to print gcode sliced for a MK3 printer the MK3.9 tells me that it will run in "MK3-compatibility mode".
However, G90 resets the extruder to absolute positioning, which does not happen on MK3 resulting in a failed purge line.

**How to reproduce**
Print the attached G-code on a MK3.9 or MK4 printer.
Accept the compatibility warning stating that the file will be run in MK3-compatibility mode.
During printing of the purge line, only half will be printed followed by a 21 mm retraction of the filament - causing a failed first layer of the print.

```gcode
M83  ; extruder relative mode
G90 ; use absolute coordinates
G28 W ; home all axes without mesh bed leveling

; several lines omitted

; purge line
; several lines omitted
G1 X240.0 E25.0 F2200.0
; several lines omitted
G1 X55.0 E25 F1400.0
; several lines omitted
G1 X5.0 E4.0 F1000.0
```

**Expected behavior**
  G90 in MK3-compatibility mode behaves the same as G90 on a MK3 printer - not resetting the Extruder stepping mode.

**G-code**
  https://drive.google.com/file/d/172TxwGLs-DS_bLN3n-HEgjLDg-ojv_xd/view?usp=sharing

**Crash dump file**
  N/A

**Video**
  N/A

bkerler, 2024-03-27:
Just to fully understand the issue: 
1. I assume that you have homed all axes with mesh bed leveling before you intent to print in mk3 mode and you just use G28 W to prevent the mbl to be done again ? 
2. The G90 should issue to use absolute coordinates, so if you say the MK4 does that, but the MK3 doesn't do that, isn't then the MK3 G90 gcode to blame which worked incorrectly before ?
Or do you mean that the MK4 G90 gcode does reset the x,y,z homing values ?

thomasloven, 2024-03-27:
I did not slice the file, so I cannot answer about the intent.

On MK4 G90 sets all axes (X, Y, Z, E) to absolute coordinates.
On MK3 G90 sets absolute coordinates for  X, Y and Z, but not E.
While I agree this is arguably incorrect behavior, it **is** documented:
[image]
https://help.prusa3d.com/article/prusa-firmware-specific-g-code-commands_112173

In compatibility mode, E is also reset - as it is in default operation on MK3.9/4 (and, I assume, 3.5).

bkerler, 2024-03-29:
Thanks, that's exactly what I needed to know. I will have a look if I can see what needs to be changed :)

thomasloven, 2024-03-29:
Oh wow. I honestly wasn't expecting that resolution due to the future maintenance overhead.

bkerler, 2024-03-29:
Happy to help wherever I can. However the Prusa Devs have to review it first.

danopernis, 2024-07-08:
This is scheduled for next minor (not patch) version of the firmware.

danopernis, 2024-08-22:
New version of the firmware 6.1.2 should contain the fix, can you test that?

thomasloven, 2025-02-24:
I am unfortunately unable to test this at this point.
I will assume it is fixed.
Thanks!

closed 2025-02-24 by thomasloven