prusa3d/Prusa-Firmware-Buddy #3850: [FEATURE REQUEST] MK3.5 To Show Sheet Profile in Footer UI

Sainx47, 2024-03-22:
**Printer type** - [MK3.5]
**Optional upgrades** - [Steel Sheet Profile in Footer UI]

Feature Request for MK3.5

1. Steel Sheet Profile on the Footer UI.
2. Renaming steel sheet profiles.

The ability to have an option to choose "Steel Sheet Profile" as a UI item to be shown in the footer.

Currently the selected steel sheet is hidden behind the control menu. Someone could easily forget which sheet is selected and print on a different one causing first layer issues. Having it shown in the footer will prevent such issues, that will lead to inaccurate z offset and failed prints.

Also, there is no option named Satin. only Smooth 1 & 2, Textured 1  & 2, Custom 1 to 4.

It would be great to rename the custom to something like PET Carbon. Or at least have Satin and PA Nylon sheet just like MK3S+

maxkueng, 2024-03-23:
Duplicate of #3789?

kajeq, 2024-03-26:
I second this feature request. This would be a big improvement.

ChrisHeerschap, 2024-03-27:
Not a duplicate of 3789 that's just adding the sheet name to the footer, which is a great idea, but I also really want the ability to rename sheets. Had it in the MK3 and used it and assumed it was there in the MK35 only to discover it's not there.

tomekmgr, 2024-04-05:
I'm just finished with an update and after initial run I realize that this feature is missing. 
You can do it, but too deep in menu. Having info and change option on main screen - please add this.

kvtorp, 2024-04-07:
I have been looking around for it, but there seems to be no way of knowing which sheet is currently selected. This is a necessary feature!

anderl1969, 2024-05-06:
I second this too. After upgrading to 3.5 yesterday, I already missing this feature a lot!

LukeShootsThings, 2024-05-12:
Commenting to show support for adding this functionality.  I have a MK3.5 and would love to be able to see which sheet profile is selected right from the footer.

CZDanol, 2024-05-16:
Hey, closing as a duplicate of #3789 . This doesn't mean we won't do this, we're only aggregating issues that are about the same thing.

closed 2024-05-16 by danopernis

IanAdd, 2024-07-29:
Awfully, confused here. This issue, well two enhancement requests, are titled as reuests for the Mk3.5. This suits me. It seems to have been consolidated into #3789 which does not mention Mk3.5, only mini, Mk3.9 and Mk4.

In truth this is a bug/oversight by the developers since the Mk3.5 is an UPGRADE from the Mk3 which had the rename feature.

CZDanol, 2024-07-29:
@IanAdd I don't get it, the #3789 mentions Mk3.5 on multiple places. And Mk4 and Mk3.9 do not have sheet profiles, Anyway, don't worry, all our printers share a firmware platform, so when we implement it, it will be implemented anywhere.

IanAdd, 2024-07-29:
> @IanAdd I don't get it, the #3789 mentions Mk3.5 on multiple places. And Mk4 and Mk3.9 do not have sheet profiles, Anyway, don't worry, all our printers share a firmware platform, so when we implement it, it will be implemented anywhere.

Thanks, It was the link to github when you click on 'scheduled' that left me concerned.

Any idea of a date of release of updates?

CZDanol, 2024-07-29:
> Thanks, It was the link to github when you click on 'scheduled' that left me concerned.
>
> Any idea of a date of release of updates?

It should get out on the next non-patch release. I'm not at liberty to share dates though, sorry.