prusa3d/Prusa-Firmware-Buddy #3851: [BUG] MK3.5 Print Area Not Centered on Build Sheet

magistersmax, 2024-03-23:
**Printer type** - MK3.5

**Printer firmware version** - 5.2.2

**Original or Custom firmware** - Original

**Optional upgrades** - Revo (but shouldn’t affect this issue)

**USB drive or USB/Octoprint** - USB Drive

**Describe the bug**
The print area on the MK3.5 is not centered in the indicated print area on Prusa steel sheets.  It is offset a few mm to the right (+X) and front (-Y).  The issue can be seen in this image: https://i.imgur.com/lpF17Sm.jpeg

I believe this may have to do with changes to Z bracket/X axis/Y axis geometry between the MK3S and MK4, since the MK3.5 uses the older brackets it may not have been properly accounted for.

I also believe that this exacerbates the post-5.2.2 issue of having poor bed leveling along the right side of the bed.  The SPINDA is pushed  further to the right (over the edge of the sheet) than expected, resulting in poor bed leveling performance.  See this reply in #3772 for a more full description of this symptom: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3772#issuecomment-2014876171

**How to reproduce**
Print a 250x210x0.2 mm rectangle on the build plate.  Time can be saved by cutting out the scented of the rectangle.

**Expected behavior**
The print should be centered.

Prusa-Support, 2024-04-02:
Thanks for reporting.

I seem to understand that the offset is reproducible even on Prusa MK3 priners and it may exist in a different extent on every combination of printer and sheet.
The graphic sings on the Spring Sheets are for reference only and not an accurate representation of the print area. 
The true (0,0,0) coordinaes may change from printer to printer and, in case of severely skewed primer geometry, the print area may even be slightly reduced.

I'd like to know if the X-axis actually allows a movement from 0 (±1) to 250 (±1) and if the full size print actually 250 mm wide (in Prusa Slicer and at the end of the print).
So far it may not seem to be a necessarily concerning issue but it may be interesting to see and compare the result on several other MK3.5 and MK3 printers to see if there is offset amount is more consistent than expected across the printers.

Michele Moramarco
Prusa Research

magistersmax, 2024-04-02:
I’m tied up with work for a while, but when I get a chance I’ll print a full first layer and get some measurements.  It might be this weekend, maybe next weekend.

It does look like there’s a number of people on Reddit experiencing the same X shift and Z offset issues along the right edge of the bed.  At this point I’m not sure if the root cause of the poor leveling along the right edge is due to the printable area being shifted as described in this bug, or if the mesh bed leveling parameters need to be adjusted so the probe stays a little farther away from the edge of the bed.

magistersmax, 2024-04-07:
Here are some photos.  The actual size of the first layer is 250x210 mm, the print is a hair under 1 mm away from the right side of the build plate.
[image]
[image]
[image]
[image]

BlueFyre, 2024-04-07:
That looks the same on my machine as well. Also, if you watch the mesh probing it probes really far right on that edge (looks like it's off the edge really) and then too far towards the front. This seems to go hand in hand with the way it's printing

mtippett, 2024-04-19:
> I share same observations on every printer I've upgraded to MK3.5 - both
> original and bear frame.
> I don't care much about print area being shifted, as long as it fits, but
> PINDA probing so close to the edge gives incorrect readings causing
> excessive (and completely unnecessary) compensation.
> For bear frame simple workaround is to shift whole bed to the right by
> 1-2mm, but stock frame would require software solution.

Can confirm this.  The PINDA on a MK3.5 will be right on the edge, IIUC,
this would provide a lower strength signal, and so the PINDA gets closer to
the steel.  This creates a virtual dip on the right side of the sheet.  I
can confirm it is still in the 6.0.0 firmware as well.

My workaround is to have the steel placed off-center to ensure the PINDA is
placed correctly.  The MBL is sufficient, although I haven't confirmed with
a full-steel 1st layer test.

IIRC, I believe that there is configurable settings for sensor to nozzle
distances.   I would assume that for MK3S/MK3.5 this is simply incorrect
and is assuming that the same location as per the nextruder.

m-kozlowski, 2024-04-19:
Sensor offset seems correct
https://github.com/prusa3d/Prusa-Firmware/blob/d22245b4823f510390e288b53dc5fdfaf4f89035/Firmware/variants/MK3S-E3DREVO.h#L459-L461
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/0de68b877b7304b770b128e1d8e9c689e4a19492/include/marlin/Configuration_MK3.5.h#L949

It looks more like whole bed is off, exactly by 1mm.
And that would match:
https://github.com/prusa3d/Prusa-Firmware/blob/d22245b4823f510390e288b53dc5fdfaf4f89035/Firmware/variants/MK3S-E3DREVO.h#L63
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/0de68b877b7304b770b128e1d8e9c689e4a19492/include/marlin/Configuration_MK3.5.h#L1111

mtippett, 2024-04-19:
Ah.   I noticed the 'MIN_PROBE_EDGE' value, which is set to zero.    I
wonder if it is calculating the probe point as being in the edge (0).
 Where it might need to be a few mm away from the edge.

russdogg, 2024-04-25:
just wanted to chime in to report I'm also having this issue with the mk3s+ to mk3.5 upgrade, 6.0.0 Buddy firmware. While printing the 3x3 calibration print: on the left side of the bed the nozzle was too high, on the right side nozzle was almost scraping the steel sheet. Even bed level correction couldn't fix it. I finally undid those changes and tried moving the sheet over 10mm to the right (so it was really close to the power supply) and running the bed leveling. Then after it was done but before the print started, moving the sheet back to its normal position. With that I got a perfect calibration print.

CoerteVoorhees, 2024-04-26:
Yep, I'm experiencing the exact same as russdogg. First Layer Calibration only works when physically shifting the plate to the right until it is nearly toughing the frame (for this I have to rotate the steel sheet 180 degrees, as the alignment pins at the back of the heat bed obviously prevent it from shifting, as intended). It seems to be that the pinda is just too far to the right side of the build plate, and missing the first column of 7 x 7 measurements, which throws off the entire mesh dramatically. Perhaps shifting the right-most probe location to the left by 5 to 10 mm would fix this in our cases.

PrintTheWind, 2024-04-29:
I'm having this issue too.  To me, it looks like the PINDA probe locations are wrong.  It looks like the nozzle is moving to the spot the PINDA should be (This results in the probe being halfway off the bed when probing the right side).  I suspect that the firmware is telling the printer to probe as if it is a MK4 rather than an MK3.5

SnowB0und14, 2024-05-01:
I’m also having the same problem.  The right side is now squishing slightly more.  The probe does look to be over the edge when checking for level.

Prusa-Support, 2024-05-04:
Thank you all for sharing your precious findings and this seems to be a common limitation on many MK3.5 based on your reports.
I'll make sure these results reach our developers.

Michele Moramarco
Prusa Research

ProstMeister, 2024-05-20:
Hello, I'm having the same problem too, on firmware 6.0.0 (my printer runs on 6.0.1 now, but after the latest FW upgrade I've never tried a 1st level calibration yet). This bug caused the printer to damage a brand new smooth steel sheet, as the nozzle squashed the filament so bad that it eventually hit the sheet damaging it. For now, instead of using the 1st layer calibration functionality, I'm adjusting the Z-offset while printing an 80x80x0.2 mm square in the center of the heatbed, as suggested by Prusa Support.

ClaGre, 2024-05-28:
Hi to everyone; well, turn the sheet of 180 degrees, move to the maximum right and first layer calibration is perfect.
Change the sheet metal, make a new calibration and perfect again.
Maybe in the next release is better to stop the calibration grid 10mm less (from the right).[image]

Mercur13, 2024-05-30:
I also have the same problem with my MK3.5. The Z-layer test always fails on the left side because the filament doesn't stick there. On the right side, however, the filament is more likely to be squashed.
I'm surprised that this bug is still present in firmware 6.0.1, even though it was reported at the end of March and confirmed by many users since then.
This unfortunately confirms my view that the MK3.5 is treated very poorly by developers and only the most glaring errors are fixed, if at all!
It seems as if there is only active firmware development for the MK4.
And since there is currently no way to update from the MK3.5 to the MK4, the only way to get out of this dead end is to sell the MK3.5. Goodbye Prusa!

PrintTheWind, 2024-05-30:
I can also confirm that this bug is still present on the newest firmware (6.0.1)

OopAck, 2024-06-02:
I am having the same issue and can not get it level on the right with my Mk3.5
I am also running 6.0.1

GalacticGreg, 2024-06-03:
Is this old bug being fixed? It damaged a brand new steel print sheet and is something that should have been caught BEFORE the mk3.5 was released. Come on Prusa!

ProstMeister, 2024-06-03:
It's been promoted to "Known bug", but the issue is still there. I damaged a brand new smooth steel sheet too. For now I'm doing the 1st layer calibration using the @ClaGre method, zero issues so far. I wish I knew this workaround _before_ damaging my new steel sheet.

OopAck, 2024-06-03:
Yes I have damaged a plate on the right side due the bed issue.
What is ClaGre Method?

ProstMeister, 2024-06-03:
> Yes I have damaged a plate on the right side due the bed issue. What is ClaGre Method?

You rotate the steel sheet 180° (with the back notch towards the front of the printer), move it to the right as close as possible to the PSU, and do the 1st layer calibration.

Before you print, remember to position the steel sheet as normally intended.

peregrines2, 2024-06-07:
6.0.2 released today and this issue STILL is not fixed.
I damaged a sheet due to this and spend many hours trying to find the cause.
Why didnt I just by Bambu Lab instead of doing this upgrade? :(

PrintTheWind, 2024-06-07:
You'd think this would be a fairly simple issue to fix

CoerteVoorhees, 2024-06-08:
@Prusa-Support I'm sure you can feel the tension/frustration building around this issue. We're all here because this bug in the 3.5 upgrade firmware has led to either damaged hardware or an inability to use the product as it was marketed and sold to us. This is not a minor issue for us. Could you kindly give us some feedback or a brief update on this status? Is the dev team aware of this? Is a fix in progress or planned for an upcoming release? We have been troubleshooting this issue and offering solutions (patiently and for free) for several months now.. I think maybe it's time for a little feedback from your side. Thank you in advance.

danopernis, 2024-06-08:
Hi, I am a developer, not a support, but here it goes:

This is a known issue, tracked internally as BFW-5606, planned for the next patch version. We figured it would be better to release the patches more often in smaller doses and unfortunately this one just didn't make it to 6.0.2

All the releases are subject to testing before they are published. As frustrating as it may be for you, we are not going to compromise our development process and risk introducing a bug somewhere else. For example, there are concerns about different PSU dimensions that some of our consumers have and we want to make sure that the fix doesn't affect them.

Thanks for understanding and hope this clarifies the situation a bit.

CoerteVoorhees, 2024-06-08:
Thanks for the update, that's great news! Maybe I speak for others in this thread when I say that getting an update and such information is really a big relief. It can be quite frustrating to be left in the dark for an extended period of time, so thank you for shedding some light on the situation. Good luck with the rest of the implementation!

ClaGre, 2024-06-15:
> @Support: Has PRUSA a refund for those customers, who crashed their Steelsheet caused by this issue !

Who should I write to to receive the crashed Steelsheet? :-D
My right side has several marks that were made before I found the solution to move the Steelsheet, although honestly I just need to avoid the prints getting to the marks.

flymac, 2024-06-19:
Same issue here after an upgrade to MK3.5. Also trashed a heat bed because of this.... Firmware 6.0.2.

artistassembly, 2024-06-22:
**MK3S+ to MK3.5 upgrade.  Firmware 6.0.2.**  

After many hours of trying to figure out why I couldn't get a good 1st layer I found this thread.  I even went back to the silicone bed mod.  Manual measurements from the z axis to the bed indicate the bed is pretty level, less than 0.06mm variance from the highest to lowest point.  

Unfortunately, when I try to print something close to the full width of the bed, the right side is too close (filament scraped by the nozzle) & the left side is to high (filament not adhering to the bed).  It appears this bug causes errors with the bed leveling mesh even if the bed is flat.  This effectively makes the right third of the bed unusable for printing.

I hope the developers make this a priority to fix.  I went from a great MK3S+ with a nice 250mm x 210mm print area to a frustrating MK3.5 with a usable 170mm x 210mm print area.  I am unable to complete projects due to this print area restriction.  I guess I can procure some aluminum & kapton tape & roll back to a MK3S+.

artistassembly, 2024-06-23:
I tried @ClaGre suggestion of moving the steel sheet to the right.  This does yield quite an improvement with the auto bed leveling, but I suspect something else is going on as well.  My left side is still low & if I adjust the bed height manually the auto bed leveling appears to compensate for it, leaving the left side still to low.  Perhaps this is due to the heated bed magnets not being in the expected location & throwing off the compensation for them.

nickduncan7, 2024-06-24:
> I tried @ClaGre suggestion of moving the steel sheet to the right. This does yield quite an improvement with the auto bed leveling, but I suspect something else is going on as well. My left side is [...]

If yours is scraping on the left side (i.e., closer to the board and not the PSU), I worry something else may be going on. My bed level is nearly perfect on my MK3.5 except for the right side, purportedly from this specific issue

BlueFyre, 2024-06-24:
FYI I just saw "PINDA off-bed probing repaired (MK3.5)" in
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.3

nickduncan7, 2024-06-24:
The fortunate thing about having lines on the right side of the build plate is I can see the first layer calibration is now visibly moved ~1mm to the left, and I think I can say any visible digging into the bed on the side adjacent to the power supply appears resolved on my end with 6.0.3. Thanks, Prusa!

Mercur13, 2024-06-24:
It's too late for me. 
I've given up since the bug fixes on the MK3.5 firmware are going so slowly. 
I sold my MK3.5 for less than it was worth and intend to switch to MK4 or another manufacturer. 
We'll see...

danopernis, 2024-06-24:
As you already noticed, 6.0.3 is out and it should help with this issue. We will still appreciate your feedback on this and other issues. Sorry about fixes taking too long sometimes.

GalacticGreg, 2024-06-24:
6.0.3 fixed this for me. No bed level correction needed anymore.

PrintTheWind, 2024-06-24:
I'm traveling for work this week, but I'm excited to get this fixed as soon as I get back!

OopAck, 2024-06-24:
addressed here too

russdogg, 2024-06-24:
Just wanted to add that my issue is solved after updating to 6.0.3. Thank you team! Pinda sensor on the right side is now completely above the steel sheet, resulting in a more accurate bed mesh. My calibration print can use some more tweaks, but honestly I tried so many different things trying to fix this originally (messing with bed level correction settings, tightening + retightening the standoffs in different patterns, adjusting z-offset to height that works with the issue) that I need to start over on calibrations to get a perfect first layer. 
[image]

artistassembly, 2024-06-26:
While Firmware 6.0.3 appears to have fix the PINDA probe issue things still appear to be _off_.  If this needs to be a new bug report please let me know,

When printing, the nozzle height is too far from the left side of the bed & too close to the right side.  Bed Level Correction can compensate for this, it's about -5 μm on the left & +20 μm on the right.  I use the silicone mod & raising or lowering the bed manually does not produce a change in the print.  This leads me to believe the Auto Bed Leveling is calculating the distance from the bed consistently but not correctly across the bed.  

The MK3S+ board & firmware compensated for the magnets in the heated bed.  Does the MK3.5 firmware have this compensation?  If yes, perhaps the MK3.5 firmware is compensating for the magnets but the position of the magnets in relation to the PINDA probe points are no longer aligned.  If no, maybe the magnets are throwing off the Auto Bed Leveling causing inaccuracy in the PINDA measurements.

In addition, a print centered in Prusa Slicer 2.7.4 is not centered when printed.  This can be seen in photo above this post from  @russdogg, and in the image at https://imgur.com/a/GOwtctx.  I would approximate the print is shifted to the right by 4mm & to the front by 5mm.  This prevents Prusa Slicer from warning when objects will be printed off of the bed & utilizing the full print area.

danopernis, 2024-06-26:
> When printing, the nozzle height is too far from the left side of the bed & too close to the right side. Bed Level Correction can compensate for this, it's about -5 μm on the left & +20 μm on the right.

Bed level correction is exactly the feature to be used in such a case.

> The MK3S+ board & firmware compensated for the magnets in the heated bed. Does the MK3.5 firmware have this compensation?

There is no magnet compensation in Buddy firmware. From what I understand, ABL is only performed with steel sheet on and this should sufficiently shield off the magnetic field.

> In addition, a print centered in Prusa Slicer 2.7.4 is not centered when printed. 

I will cite my colleague here:

> The graphic sings on the Spring Sheets are for reference only and not an accurate representation of the print area.
The true (0,0,0) coordinates may change from printer to printer and, in case of severely skewed primer geometry, the print area may even be slightly reduced.

I agree that this issue is becoming cluttered. Therefore I am closing this one, as it is related to the Pinda being outside of the bed sheet, which was fixed. Feel free to open another issue.

closed 2024-06-26 by danopernis