prusa3d/Prusa-Firmware-Buddy #3852: [BUG] MK3.5 probing slightly off right edge of the bed

Rods87, 2024-03-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK3.5]

**Printer firmware version** - [5.2.2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  USB/Octoprint

**Describe the bug**
  During mesh leveling, PINDA extends about 2mm+/- beyond the right edge of the bed.  This results in incorrect bed mesh & too much squish when printing close to the right edge of the bed. Z homing point is about 3mm+/- to the right of the printed circle on the heated bed. 

**How to reproduce**
  Perform mesh bed leveling

**Expected behavior**
 All probe points should be fully on the bed and normal first layer height should be possible close to the right edge of the bed.
**G-code**
 N/A

**Crash dump file**
  N/A

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

**Video**
  Please attach a video. It usually helps to solve the problem.

Rods87, 2024-03-26:
N/a

closed 2024-03-26 by Rods87

PrintTheWind, 2024-04-29:
I'm having this issue on my MK3.5 on firmware version 6.0.0.  How did you fix this?