prusa3d/Prusa-Firmware-Buddy #3854: [ENHANCEMENT]  While waiting for a docked tool to heat, show that tool's status

djdelorie, 2024-03-25:
**Printer type** - XL multi-tool

During normal printing, there's a point where the tool head is returned to its dock while it heats up.  However,  while waiting, the bottom status line on the LCD shows the "no tool" icon, so you can't tell what it's waiting for, or how hot the nozzle is.  Instead, if you're waiting for a tool, that tool's status icon (including temperature, and blinking) should be displayed on the status line.

I.e. the status line should always show you what you're *waiting for*, not just what's *loaded*.

djdelorie, 2024-06-12:
Commenting to remove the stale label.  I still think this enhancement would be very useful for users, although I understand if it takes a while to get to...

BrienKing, 2024-07-01:
I agree.  There is room vertically on the left to show the state of ALL the tools.  Might not look as pretty, but I'm more interested in the functionality of rather than whether everything is symmetrical, etc...

djdelorie, 2025-11-17:
ommenting to remove the stale label. I still think this enhancement would be very useful for users, although I understand if it takes a while to get to...