prusa3d/Prusa-Firmware-Buddy #3855: [FEATURE REQUEST] tool head "hot" indication

djdelorie, 2024-03-25:
**Printer type** - XL multi-tool

It would be useful (and safer) if the green LED on the front of the tool heads could somehow indicate which tool heads are "hot" (i.e. dangerous to touch).

kvtorp, 2024-04-07:
Yes! I have also tried to burn my fingers because I mistakenly thought that toolheads would cool down after unloading filament

wafflecart, 2024-04-07:
> Yes! I have also tried to burn my fingers because I mistakenly thought that toolheads would cool down after unloading filament

Yea I noticed this behaviour is a bit backwards, load filament and it cools down to 170c. Unload keeps it at full temp eg 215c for PLA. Doesn’t really make sense. It’s like they assume you will load filament again after unloading.

djdelorie, 2024-06-12:
Commenting to remove the stale label.  I still think this feature would be very useful for users, especially as it's safety related, although I understand if it takes a while to get to...