prusa3d/Prusa-Firmware-Buddy #3856: [BUG] XL Alpha 6.0.0 Fails Tool Offset side-probing, Downgrading to v5.1.2 restores proper offset probing.

Mykal451, 2024-03-25:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - XL (5 tool heads)

**Printer firmware version** - 6.0.0

**Original or Custom firmware** - n/a

**Optional upgrades** - n/a

**USB drive or USB/Octoprint** - USB Flash Drive

**Describe the bug**
Performed 2 Tool-Head to 5 Tool-Head upgrade. When running the tool offset calibration, Extruder 1 performs the initial downward tap to center of calibration pin, the bed then drops some distance and the extruder starts to side probe the pin, the bed is too low and the nozzle moves right to left over the top of the calibration pin and the printer then errors-out with "Pin Not Reached", see image.

**How to reproduce**
Prusa XL with multiple tool heads running firmware alpha v6.0.0, run tool offset calibration.

**Expected behavior**
  Bed does not drop so low that nozzle does not side tap calibration pin.

**G-code**
n/a

**Crash dump file**
n/a

**Video**
  Please attach a video. It usually helps to solve the problem.
[image]

wafflecart, 2024-03-26:
Same as #3797 please add your findings to there.

dpark12300, 2024-03-30:
Same as my #3873 3873

wafflecart, 2024-03-31:
> Same as my #3873 3873

You opened that 17hrs ago.. #3797 is the ticket for all this it was opened a month ago.

danopernis, 2024-04-30:
This seems to be a duplicate of #3797 which was fixed in the final version 6.0.0 of the firmware. @Mykal451 can you confirm the issue is no longer present in 6.0.0?

closed 2024-05-31 by danopernis