prusa3d/Prusa-Firmware-Buddy #3859: [BUG] MK3.5 Firmware 5.2.2 - Can not transfer a bgcode file to the printers USB drive over the network

dmyers7, 2024-03-28:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK3.5]

**Printer firmware version** - [5.2.2.]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [MMU3]

**USB drive or USB/Octoprint**
  USB Drive - The drive that was supplied with my MK3.5 upgrade

**Describe the bug**
 Can not upload binary Gcode from PrusaSlicer 2.7.3 through PrusaLink or PrusaConnect. PrusaLink gives an error that the file is not a GCode file
PrusaConnect reports that the printer is offline; even though PrusaConnect is showing me my printer, it's telemetry, files, .....

I can remove the Prusa USB Thumb drive, plug it into my laptop and save the bgcode file directly to it from PrusaSlicer, plug it back into the MK3.5 printer and it sees it and I can start the print.

I insured that the API Key for both PrusaLink and PrusaConnect was set correctly in PrusaSlicer and "tested" the connection with no issue.

This has happened with Multi-material designs and single color designs, insensitive to input design.

**How to reproduce**
Generate a sliced design using the MK3.5 Prusa Profile and try to upload it to a MK3.5  running 5.2.2

**Expected behavior**
It worked before, the file should get stored to the USB drive and present itself in the Queue or start Printing

**G-code**
Insensitive to a particular design / code file.

**Crash dump file**
Does not crash

**Video**
N/A. Ask if you really need one.

dmyers7, 2024-03-31:
I can confirm this to be the same problem in the MK3.5 V5.2.2 Firmware; shorten the name and it uploads.

godlike64, 2024-04-10:
Came here after finding [this issue](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3039). Just tried sending a .bgcode with a long filename (`longer-front-lip-version-for-monitors-with-curved-or-thick-bezels-print-in-place_0.4n_0.2mm_PLA_XLIS_3h37m.bgcode`) to my Prusa XL via PrusaLink (via normal drag-and-drop on the browser) and it throws this "415" error (something about "Not a GCODE"). Maybe it's more an issue of PrusaLink than a specific printer model?

danopernis, 2024-04-17:
Hi, thanks for the report. I just tried this in the newest 6.0.0 firmware and uploading `.bgcode` file works.

greggjaskiewicz, 2024-04-30:
I'm getting the same issue on my MK4, running 6.0 and using prusa slicer 2.7.4 on macOS.
Uploading file onto USB stick (binary code) and running it works fine. So bit confused here.
[image]

closed 2024-07-09 by github-actions[bot]

machacekch, 2024-08-20:
This is still broken on 6.1.0 on the MK4S. Here's an example file name that triggers a 415 (Unsupported Media Type) HTTP error "Not a GCODE" when uploaded via PrusaLink web interface:

```
file_name_0123456789012345678901234567890123456789012345678901234567890123456789012345678901234.bgcode
```

Note that it contains no spaces or other special characters. Its length is 95.6 and shortening it by just one character to 94.6 makes the upload work. Please try reproducing the issue with this file name and close it only after you've verified that it's fixed (ideally adding a test to make sure it remains fixed).

danopernis, 2024-08-20:
Duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3039