prusa3d/Prusa-Firmware-Buddy #3864: [BUG] 6.0.0-RC1 - Wifi doesn't work

Gyrxiur, 2024-03-29:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [6.0.0-RC1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [none]

**Describe the bug**
  After updating to 6.0.0-RC1 Wifi have worked, but only till restart, then it says No AP. Only way to make it work is factory reset, but it again works only till restart. Also, in czech language, you have typo, instead of Pokus o připojení you have Pokus o piřpojení ;c)

**How to reproduce**
  Update firmware to 6.0.0-RC1 and restart.

**Expected behavior**
  Working wifi every time, as it was before updating.

**G-code**
  N/A

**Crash dump file**
  N/A

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

**Video**
  N/A

Gyrxiur, 2024-03-29:
So I've reverted to old firmware and it works, so it is bug in new firmware for sure.

bkerler, 2024-03-29:
Do you have a prusa-printer-settings.ini file on your stick ? Sounds like a ini parsing issue maybe, trying to sort that out.

Gyrxiur, 2024-03-29:
> Do you have a [prusa-printer-settings.ini](http://prusa-printer-settings.ini) file on your stick ? Sounds like a ini parsing issue maybe, trying to sort that out.

Yes, and when I load settings, it asks to delete files, so I confirm it. And even when I try to make new file and load it again, it won't connect. Older firmware connects ok with same file, so no mistakes in that file for sure.

H0Z0, 2024-03-29:
Interesting. I tried to reproduce the error, but after setting up the network, my MK4 connected to the Wi-Fi network every time. It also works after switching off or disconnecting the power cable or pressing the reset button. I also deleted the .ini file from the usb drive, so it can't even read from there. Both PrusaLink and Prusa Connect can connect to the printer.

salmander, 2024-03-29:
I confirm that I also experienced this. However, I have flashed the RC1 firmware and restarted the printer a couple of times and now it seems to be working. Another issue I experienced is when I am using binary code (bgcode), it can't seem to download the file from Prusa Connect. I tried this a few times. However, changing the legacy gcode in PrusaSlicer worked the first time.

Gyrxiur, 2024-03-30:
So, I have upgraded to 6.0 again, and made some observations. Wifi works, if I have machine turned of for at least 5 minutes, when I reboot it, it stops working, and even if I turn it off for a minute, to really power off wifi module, that doesn't help. It has to be off for at least 5 minutes, so it looks like there is something wrong with combination with my Huawei WiFi AX3 router it seems. But strange is, with older firmware it works, downgrading fixes it. I've even tried to turn on backup wifi 5 network in router and connect to it, but that doesn't help.

Gyrxiur, 2024-04-16:
Updated to final 6.0, problem persists. Also that typo not fixed.

Gyrxiur, 2024-04-16:
OK, so, I had router in WPA3/WPA2 mode, when I switch it to WPA2 only, it works. So new firmware has problem with WPA3, while older one works with it enabled.

danopernis, 2024-04-16:
@Gyrxiur Thanks for figuring out it is an issue with WPA3. Which version were you upgrading from? AFAIK the WiFi is entirely driven by ESP8266 so it must be caused by one of the commits in https://github.com/prusa3d/Prusa-ESP-Nic-FW/commits/master/

Gyrxiur, 2024-04-16:
5.1.3 was working well, all 6.x have this issue.

danopernis, 2024-04-16:
Unfortunately there is no 5.2.x release for MK4 so we can't easily confirm if it is caused by https://github.com/prusa3d/Prusa-ESP-Nic-FW/issues/15 or the earlier changes. If there are any MK3.5 users who could help, that would be great.

I will try to reproduce this on some other router.

Gyrxiur, 2024-04-16:
Wouldn't it be possible to compile Prusa signed 6.0 firmware with various versions of ESP firmware, so I could test them, to find out, when it became broken? I am willing to test it, but I don't want to brake asterisk on motherboard and ruin my warranty.

jaromirk, 2024-04-16:
I can confirm the same issue. My AP is Mikrotik that allows both wpa2 and wpa3.

danopernis, 2024-04-17:
@Gyrxiur thank you for generous testing offer, but I think that now we have enough leads to reproduce this. Internal ticket for reference: BFW-5393

For anybody experiencing the issue, you can work around this by disabling "WPA3/WPA2 Transition mode" until the proper fix is made.

dzuelke, 2024-04-17:
Fairly sure this is PMF related.

https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/e8ac2cf49d05751efa09e2cb0abd3bd04dc70228 added support for PMF, but maybe `required` now somehow defaults to `true`?

Or it kicks into WPA3 mode this way, but there's a bug there? https://docs.espressif.com/projects/esp-idf/en/v5.0.6/esp32/api-guides/wifi-security.html#setting-up-wpa3-personal-with-esp32

Note that https://docs.espressif.com/projects/esp-idf/en/v5.0.6/esp32/api-reference/network/esp_wifi.html#_CPPv417wifi_pmf_config_t indicates `capable` no longer has any effect on recent versions, this is confirmed by https://docs.espressif.com/projects/esp-idf/en/v5.0.6/esp32/api-guides/wifi-security.html

WPA3 is specced to require PMF, while for WPA2, it is optional. https://docs.espressif.com/projects/esp-idf/en/v5.0.6/esp32/api-guides/wifi-security.html has some details.

In general, this is a common cause for trouble with IoT devices - WPA3 is often not supported, and WPA3/WPA2 transitional mode with optional PMF for WPA2 then has people reporting "my printer/vaccum/sensor/camera/… isn't connecting anymore".

Places like the UniFi community forums are full of these reports. Router/AP vendors often try to fix problems for one type of product, and in the process, break support for another, because these simple WiFi chipsets are often fairly glitchy.

The usual recommendation is to run a separate IoT WiFi that uses only the 2.4 GHz band and WPA2 Personal for maximum compatibility.

danopernis, 2024-04-22:
Update:

I tried to reproduce this on the network with enabled WPA2/WPA3 Transitional mode and the printer connects just fine. This seems to be related to the particular router you are using. Maybe there is some ESP8266 configuration I could tweak, but I have no way of verifying if that tweak solves the issue for your particular router.

As pointed out by @dzuelke this is likely to be caused by https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/e8ac2cf49d05751efa09e2cb0abd3bd04dc70228 (or more specifically, https://github.com/prusa3d/Prusa-ESP-Nic-FW/commit/ed2613339dcfdc007672ca1eea72ada36751f684 and https://github.com/prusa3d/Prusa-ESP-Nic-FW/issues/15). We do not want to revert this because it enables WPA3 which is desired. For future reference, I am also including link to [discussion at reddit](https://old.reddit.com/r/prusa3d/comments/1c8d9a4/mk4_supports_wpa3_wifi_security/) where they talk about WPA3 finally working.

For now, the best workaround is either use WPA2 or WPA3 on your router.

If there is anybody who experiences this issue and is skilled enough to build https://github.com/prusa3d/Prusa-ESP-Nic-FW from sources and flash their ESP-01S, I would really appreciate if you could try tweaking the configuration. Right now the printer does not flash ESP firmware as long as the protocol version matches, so experimenting should be easy.

MarcusPocus66, 2024-04-23:
Problem reproduced and confirmed with Prusa technicians.  The bug is there with 6.0.0 but not with 5.1.3.

Prusa-Support, 2024-05-07:
Thanks for reporting.

As our developer mentioned, we are already working on this. For now, a potential workaround may be setting the router fixed on WPA2 (or WPA3 in a few cases),  without automatic transitioning. Please let us know your further findings.

However, we struggle to reproduce the issue which may seem to be modem-dependent.
In case of problems like these, we would appreciate you sharing more details about the router model/specifics and settings (if you customized any). Thanks.

Michele Moramarco
Prusa Research

chinskiy, 2024-05-07:
Also faced this issue with Mikrotik hAP ax3
Can confirm that disabling automatic transitioning mode WPA2/WPA3 solves an issue like @Prusa-Support suggested

MarcusPocus66, 2024-05-07:
Faced the issue with HP / Aruba AP12 semi-commercial access points.  The only option I have is “WPA2+WPA3 personal”.

onovy, 2024-05-07:
Can confirm on Mikrotik cAP ac with "new" wifi drivers (wifi-qcom-ac). Relevant part of configuration:

```
/interface wifi security
add authentication-types=wpa2-psk,wpa3-psk disable-pmkid=yes disabled=no encryption=ccmp ft=yes ft-over-ds=yes group-encryption=ccmp name=Name wps=disable
```

tjdomsalla, 2024-05-08:
Same issue here.
Using a Fritz!Box 7590. After changing the WPA method from "WPA2 (CCMP)" to "WPA + WPA2" is seems to work now.

mitchellj, 2024-05-10:
Same issue using a Mikrotik hAP ax3

pikdum, 2024-05-12:
Running into this issue with a TP-Link AC1200 v2 running OpenWrt 23.05.3.

cbricart, 2024-05-15:
I also face this issue since upgrading the MK4 to 6.0.0
funnily enough - for exactly such reasons - my WiFi sticks to a WPA2 rather than WPA2+3-Transition configuration. However, this TP-Link-Omada also advertises PMF as "capable" by default - even in WPA2-only mode - so that matter might hint at PMF again.

Gyrxiur, 2024-06-07:
For anyone asking themselves, still not fixed in 6.0.2, don't bother switching your router settings back to WPA3 transitioning mode.

mitchellj, 2024-06-10:
Having to work around this problem by having a wifi extender plugged into the power strip under the prusa and running a RJ45 cable from the wifi extender to the printer. @Prusa-Support any update since May 7th?

CZDanol, 2024-06-12:
@pikdum "TP-Link AC1200" is not an exact model name. Can I ask you to provide the exact model name of your router?

pikdum, 2024-06-20:
TP-Link AC1200 is the model name and v2 is the hardware revision. Not sure what else you're looking for.

CZDanol, 2024-06-20:
@pikdum okay, thanks. We'll get the router and try debugging it in our office :)

jaromirk, 2024-06-20:
@CZDanol I can send you my mikrotik config if needed. Mikrotiks will be likely easier to get.

onovy, 2024-06-20:
already did:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3864#issuecomment-2098963005

kwinsch, 2024-06-29:
Also tried it with 6.0.3 on a MK4. Still not working and had to downgrade to version 5 firmware.

Gaugeforever, 2024-07-01:
I don't know if this is related to all of your problems, but I had this issue.  I resolved this by reflashing my ESP module.  There must be an ESP module firmware update they didn't mention or something.  Once I flashed this I didn't even have to redo my credentials, it connected and everything is fine.

In the future, there should probably be some UI element to warn the user that new firmware requires an ESP module firmware update as well if that is the case.  Or to do it automatically.

jaromirk, 2024-08-14:
Just noticed, that my printer on 6.0.4. is connected to both wifi and Lan. Does it work also for others?

mplzik, 2024-08-18:
I can confirm this is still an issue; Prusa MK4 firmware version 6.0.4, Mikrotik hAP ac2, RouterOS 7.15.3

jaromirk, 2024-08-18:
It was connected, I rebooted and it was no longer able to connect. Weird! 6.0.4

MarcusPocus66, 2024-08-20:
Do we have confirmation that Prusa is actively working on this bug?  On my side, I can no longer use wi-fi, so I installed a wired connexion.

Gyrxiur, 2024-08-22:
Still not fixed in 6.1.2.

bkerler, 2024-08-24:
I just tested some APs. Here are the test results so far, these worked fine with v6.1.2 on MK4 (ESP8266):
AVM Fritzbox 7590: WPA2/WPA3
tpLink Archer AXE5400 : WPA2-AES[PSK]
tpLink Archer AXE5400 : WPA3-Private
tpLink Archer AXE5400 : WPA3-Private/WPA2-AES[PSK]

Note: When switching the network security on the router, the wifi adapter has to be switched from wifi to eth and back in "Settings->Network->Active Interface", otherwise the MK4 doesn't adapt the new router security settings.

bkerler, 2024-08-24:
I assume that on the mikrotik devices, capsman is disabled or configured properly ? See [here](https://forum.mikrotik.com/viewtopic.php?t=171568) for reference.

Gyrxiur, 2024-09-11:
Still not fixed in [6.2.0-alpha1]

mbaguette, 2024-10-06:
If that can help, I have the same issue with Aruba AP-305H, on 2.4Ghz SSID with WPA3-SAE. (WPA3-Personal).
The connection goes up and down directly from the top right icon.
Switched back to WPA2-AES SSID and it worked first time, stays connected.
Tried with the firmware 6.2.0-alpha2 for my MK4S.

peterbabic, 2024-11-27:
Happens on Mikrotik `Chateau 5G ax` but not on `Mikrotik hap ac3` on Buddy 6.1.3

onovy, 2024-11-27:
Mikrotik hap ac3 is broken for me too with WPA2/3 mixed mode. Broken for months and months. Shame.

exxrxxe, 2025-05-26:
I have same issue on UniFi using WPA2/WPA3 mode

3d-gussner, 2025-08-11:
@exxrxxe I have found this information:

If you want the MK4 to connect in WPA2/WPA3 mode on UniFi:

-     Set PMF to “optional” instead of “required” in the WLAN settings.

-     Keep WPA2/WPA3 mixed selected, but disable WPA3 fast roaming (some IoT clients break on this).

-     Use only AES/CCMP cipher for WPA2, not TKIP.

Gyrxiur, 2025-08-14:
@danopernis Zdravíčko, vidím, že ste z Prahy, ja som z Martina na Slovensku. Tak píšem po slovensky. Ak som dobre pochopil, vy pracujete pre Prúšu? Ide o to, že som si kúpil nový router, lebo ma to hnevalo, ja proste chcem používať WPA3, lebo mi ide cez neho stabilnejšie a rýchlejšie net. No napadlo ma, nedalo by sa dohodnúť, že by som Vám poslal svoj starý router, keďže je to tu označené, že sa u Prúši nepodarilo replikovať problém, a od Vás by ste mi poslali za to dajme tomu dve 1 kg špulky Prúšamentu, jednu PLA, druhú PETG, farbu by som si vybral, nech nie som škodný? Vám by to zlepšilo reputáciu, keby sa problém vyriešil, keďže na staršom FW to šlo a doteraz ste nevrátili FW ESP Wifi modulu na staršiu verziu, ktorá bola súčasťou FW pred verziou 6.0, kde to fungovalo OK?

jaromirk, 2025-08-30:
@Prusa-Support enabling wpa2 is not a solution, it's a workaround. You should aim for wpa3 native support. Thx.

hdiethelm, 2025-10-22:
Nice, more than one year open and still not fixed... :-( 
So it took me several hours to figure out why my printer doesn't connect to my new mikrotik access point. Until now, the prusa is the only device having this issue.

I am using the following AP: https://mikrotik.com/product/l23ugsr_5haxd2haxd (Board from NetMetal ax)
wAP ax has similar hardware

Anyway, for other people having the same issue while prusa is not interested in buying a mikrotik AP and figuring out a proper solution:
Crapy workaround: 
Disable WPA3-PSK on 2.4GHz (Only WPA2-PSK active)

Nice workaround:
Add a new SSID like PrusaWPA2Only: Wifi->New: set SSID/WPA2-PSK/Passphrase and master to the 2.4GHz interface
Add to bridge: Bridge->Ports->New and add the newly created wifi interface to bridge

For prusa, it would be easy to add a wifi option to only use WPA2 as long as WPA3 doesn't work everywhere and it would have saved me several hours.

hdiethelm, 2025-10-26:
> [@hdiethelm](https://github.com/hdiethelm) Thanks for sharing, and sorry that you had such hard time.
> 
> Have you tried [#3864 (comment)](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3864#issuecomment-3175409552)

I guess this is related to the following issue:
https://github.com/prusa3d/Prusa-ESP-Nic-FW/issues/15

In hope that prusa engineers use the information to improve the software, I did some testing:
1. Enable WPA3-PSK: Immediate connection loss
2. .management-protection=allowed -> Still no connection
3. .management-protection=disabled -> Error: management protection should be allowed or required with WPA3 & WPA2
4. Disable WPA3-PSK: Connection immediately regained

I think a lot of consumer access points disable PMF on WPA3 even if the standard requires it to be compatible with IOT devices not following the standard.

FlorentNogema, 2025-11-17:
> The MINI and MK4 use an [ESP8266](https://www.prusa3d.com/product/wi-fi-adapter-esp-01s-for-mini-and-mk4/) module while the MK4S and CoreOne an [ESP32](https://www.prusa3d.com/product/wi-fi-adapter/)
> 
> My MK4 can connect to my OpenWrt router with a WPA2-PSK/WPA3-SAE Mixed WiFi.
> 
> Please check [#3864 (comment)](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3864#issuecomment-3175409552)

Our access points doesn't have this kind of settings, it's very minimalist, we can only configure the "security" (the wpa variant). 

Is there nothing that can be done in the esp8266 firmware to improve compatibility?