prusa3d/Prusa-Firmware-Buddy #3866: [BUG] missing linefeed in serial output

smurfix, 2024-03-29:
**Printer type** - any XBuddy

**Printer firmware version** - current

**Original or Custom firmware** - Original

**Describe the bug**
The attached output demonstrates that there's at least one missing line feed in the firmware. This causes Octoprint (or whatever else is sending commands to the printer) to ignore the next line from it.

In this case it's a temperature status, thus it doesn't matter if one is missed, but it could well be something else.

**Expected behavior**
All serial output from the printer is terminated by a line feed.

** GCode **
```
Send: M155 S2
[…]
Send: N44 G29 P3.13*77
Recv: Extrapolating mesh... T:184.00/185.00 B:80.25/80.00 C:-30.00/0.00 X2:31.00/36.00 A:67.45/0.00 T0:22.00/0.00 T1:25.00/0.00 T2:184.00/185.00 T3:24.00/0.00 T4:25.00/0.00 T5:6.00/0.00 @:36 B@:0 HBR@:51 @0:0 @1:0 @2:36 @3:0 @4:0 @5:0 B_0_0:27.10/0.00 B_1_0:60.10/60.00 B_2_0:80.20/80.00 B_3_0:60.20/60.00 B_0_1:27.50/0.00 B_1_1:60.10/60.00 B_2_1:80.30/80.00 B_3_1:80.20/80.00 B_0_2:26.90/0.00 B_1_2:40.10/40.00 B_2_2:60.10/60.00 B_3_2:60.10/60.00 B_0_3:26.40/0.00 B_1_3:27.70/0.00 B_2_3:28.70/0.00 B_3_3:27.90/0.00
Recv: done
```

bkerler, 2024-12-14:
@smurfix I just checked the code and both M155 and G29 have a proper serial linefeed. I'm not sure to understand what the issue actually is ? Can you please elaborate more ?

smurfix, 2024-12-14:
You misunderstood the reported bug. The problem is the missing newline after "Extrapolating mesh...".
lib/Marlin/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp line 2023 in master.

bkerler, 2024-12-14:
I wrote a PR. If I do understand correctly the issue is that the Serial output is mixed due to the temperature auto report sending data while the requested gcode command hasn't finished sending its own serial data.

bkerler, 2024-12-14:
Nevertheless, I also added a new line after "Extrapolating mesh..." :)

bkerler, 2025-04-13:
I assume this issue can be closed, as the fix was merged ?

closed 2025-04-13 by smurfix