prusa3d/Prusa-Firmware-Buddy #3871: [BUG] Enabling stealth mode disables progress on screen

NotSwedishBacon, 2024-03-30:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 6.0.0-RC1+14732

**Original or Custom firmware** - Original

**Optional upgrades** - N/A

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  When enabling stealth mode, the progress switches to zero % and remains there for the duration of the print. Remaining time is also N/A. 

**How to reproduce**
  Turn on stealth mode either before or during a print

**Expected behavior**
  Progress bar remains and increases as expected, remaining time should be remaining time in normal mode + an offset to account for stealth mode

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.
[mini-jeux-puissance-420231209-292645-mxvqud.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/14812246/mini-jeux-puissance-420231209-292645-mxvqud.zip)

**Crash dump file**
  Please send the crash dump file to Prusa by emailing it to reports@prusa3d.com. Sharing this file is important and helps us investigate the bug.

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

**Video**
  Please attach a video. It usually helps to solve the problem.

ErGeeBee, 2024-03-30:
Having the same issue. Also when finishing printing my printer shows 99% instead of 100%. Not sure if this is only when stealth mode is enabled. See pics.
[image]
[image]
[image]

fmorgner, 2024-03-30:
Based on a quick check, I get the feeling that the missing percentage and time is most likely an issue in PrusaSlicer. Comparing files sliced for a MK3 and a MK4, the slicer does not emit the appropriate `M73` codes for the silent mode when slicing for the MK4.

I've [attached a file](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/14813327/cube-gcode.zip) (the same model, sliced for a MK3  and a MK4 respectively), generated with PrusaSlicer 2.7.3.

Of course this is assuming that the underlying `M73` logic in the firmware is still the same between the MK3 and MK4.

H0Z0, 2024-03-30:
Did you turn on Printer settings - General - Firmware section - Supports stealth mode in the slicer before generating the g-code?

ErGeeBee, 2024-03-30:
> Did you turn on Printer settings - General - Firmware section - Supports stealth mode in the slicer before generating the g-code?

This solved it for me. Thanks!
Could Prusa enable this by default on their system presets?

H0Z0, 2024-03-30:
> > Did you turn on Printer settings - General - Firmware section - Supports stealth mode in the slicer before generating the g-code?
> 
> This solved it for me. Thanks! Could Prusa enable this by default on their system presets?

I'm glad it worked out. Maybe after the stable release there will be a default value in the slicer.

NotSwedishBacon, 2024-03-31:
Yeah I'll close this then, being as it isn't the firmware. Ticking the box in the slicer worked fine for me

closed 2024-03-31 by NotSwedishBacon