prusa3d/Prusa-Firmware-Buddy #3873: [BUG] XL moves up during offset calibration.

dpark12300, 2024-03-30:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - XL

**Printer firmware version** - 6.0.0 Alpha 1 (5.1.2 works fine)

**Original or Custom firmware** - Original

**Optional upgrades** - 2 Toolheads

**USB drive or USB/Octoprint**
USB
**Describe the bug**
  Printer moves up instead of down during tool offset calibration

**How to reproduce**
  Run tool offset calibration (not dock offset)

**Expected behavior**
  The printer moves down to tap the side of the probe

**G-code**
  No GCODE was loaded during calibration

danopernis, 2024-04-30:
This seems to be a duplicate of #3797 which was fixed in the final version 6.0.0 of the firmware. @dpark12300 can you confirm the issue is no longer present in 6.0.0?

dpark12300, 2024-04-30:
Yes, It seems to be fixed.

closed 2024-04-30 by danopernis