prusa3d/Prusa-Firmware-Buddy #3874: [BUG] Burned filament during Spool Join

Kosaro, 2024-03-30:
**Printer type** - [XL]

**Printer firmware version** - [5.1.2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [5 head tool changer]

**USB drive or USB/Octoprint**
  USB flash drive

**Describe the bug**
 Toolheads configured for spool join stay hot throughout the print, burning the filament in the nozzle and contaminating the print (see picture).

**How to reproduce**
 1. Slice long print that uses one toolhead.
 2. Load filament into toolheads 1 and 2.
 3. Set toolhead 1 as primary toolhead.
 4. Set up spool join for toolhead 2.
 5. Print for extended duration using toolhead 1 (in my case 30 hours using PLA), wait for filament to run out.
 6. Observe burned plastic coming out of toolhead 2 after spool join.

**Expected behavior**
  The toolhead(s) used for spool joining should stay cold (or sufficiently cooler) until they are needed, conserving energy and preventing the filament from burning.

**G-code**
What is the best way to share binary gcode? Will share if relevant.

**Crash dump file**
NA

**Image**
The brown layer is the contaminated layer after the spool join.
[image]

Kosaro, 2024-06-12:
Has anyone had a chance to look into this?

jhnphm, 2024-06-20:
I've also observed this happening; extruders 4 and 5 are joined, and extruder 5 just sits at 215C while waiting for 4 to run out

Kosaro, 2024-06-25:
And yes I just confirmed this issue still happens with firmware 6.0.2.

Razor412, 2024-12-12:
This is still happening with 6.2.0-alpha2+8702. Also, even extruders that aren't being used or have filament in them are also at 170º.

Kosaro, 2025-02-11:
I'm no longer seeing this issue with 6.2.0.

Razor412, 2025-02-14:
I can confirm I am not seeing this with tool heads 1 & 2 under firmware 6.2.0.

danopernis, 2025-02-14:
Thanks, closing this.

closed 2025-02-14 by danopernis