prusa3d/Prusa-Firmware-Buddy #3875: [BUG] XL XLCD Audio distorted during print

dpark12300, 2024-03-31:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - XL

**Printer firmware version** - 5.1.2+13478

**Original or Custom firmware** - Original

**Optional upgrades** - 2 Toolheads

**USB drive or USB/Octoprint**
  USB Drive

**Describe the bug**
  Audio distorts during a print on my 2T XL. Both 5.1.2 and 6.0.0 Alpha 1

**How to reproduce**
  Start a print. Make sure the sound mode is set to Loud. Move around in the menu. Compare this to when not printing.

**Expected behavior**
  The audio not being distorted

**G-code**
  Multiple G-codes.

tg73, 2024-04-04:
I've noticed this too (if it's the same thing) - specifically, if a large gcode is still uploading when the print starts, then the control panel beeper thingy sounds like it has vocal fry (low frequency modulation). My guess was that some kind of bus was saturated with the network transfer traffic and the beeper driver was starved. This might suggest a broader issue, for example if other subsystems get affected by bus saturation.

braw, 2024-04-05:
I am also experiencing this. I assumed my speaker was faulty and was going to live with it, but it sounds like I'm not alone. Beep sounds fine when printer is idle, but during a print, it is distorted or warbled.

wafflecart, 2024-04-06:
I have this issue, then I also noticed there is no sound at all in the menu when you’ve hit that button to map/choose filament or extruders on a multi material print.

Replication:
- In prusa XL upload a multi material/color print
- On interface press the button with the squiggles (mapping filament thing). Inside here there is no noise even though I’ve got it turned on in settings.

danopernis, 2024-06-12:
Hi folks, this should be fixed since v6.0.0 can you please confirm?

braw, 2024-06-12:
I am still experiencing audio distortion while printing. I'm on version 6.0.2.

CloudStrife1983, 2024-12-30:
Will this ever be fixed? The problem has been around for 9 months now.

danopernis, 2024-12-31:
Hi, this is not very high on the priority list, but I tried to bump internal ticket some more. Fix would require us to find some unused STM32 hardware timer on XL with sufficient frequency. Right now the sound on XL is handled from 1kHz timer which can't possibly work for all the tones folks would like to play. If anyone is interested in creating PR, I would be happy to merge it.

CloudStrife1983, 2025-02-24:
Where can I find out which revision I have?

ivous, 2025-02-24:
> Where can I find out which revision I have?

Menu > Info > Version Info > Buddy Board --> if 9 and higher, it is the newest, buzzer working fine. 
If lower, then the buzzer is using SW driving and you may  hear distortion when processor is busy.

Just warning, the fix is not published yet. When it will be available, this ticket will be marked as solved. Exact date of publishing is not planned yet, it will not be in the next planned revision.

closed 2025-11-24 by github-actions[bot]