prusa3d/Prusa-Firmware-Buddy #3876: [BUG] Blue Screen Crash - 5T XL

GuyH77, 2024-03-31:
Came back to the printer beeping, print seems to have completed but had a blue screen.

XL 5T - Printing on T1 0.6n, 2-5 = 0.4n

6 Alpha

USB Thumb drive

Came to a beeping printer with blue screen

[image]
[XL_Crash.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/14816176/XL_Crash.zip)

gcode and crash log attached.

michaelaswood1980, 2024-03-31:
I have had the exact same issue today. Printing fine and then as I started a new print via connect I heard the printer beeping and on inspection saw the blue screen crash.... Same firmware version.

danopernis, 2024-04-02:
@GuyH77 Thanks for your report and especially the attached crash dump. After the initial examination this looks like a deadlock involving config store mutex. I will open internal ticket and hopefully I will be able to fix this. Sorry for your troubles :disappointed:

GuyH77, 2024-04-02:
> @GuyH77 Thanks for your report and especially the attached crash dump. After the initial examination this looks like a deadlock involving config store mutex. I will open internal ticket and hopefully I will be able to fix this. Sorry for your troubles :disappointed: 
> 
> 
> 
> BFW-5302

I'm running the alpha, I know that comes with some instability. Happy to help in some small way. Thanks.

rcmaniac25, 2024-04-04:
I hit a similar issue with my single tool and sent my dump to prusa email. My BSOD has the exact same numbers on everything, so it makes me think I may have hit the same config store mutex issue.

ubermlin, 2024-04-04:
Similar issue on MK4.

CZDanol, 2024-04-13:
Hey, should be fixed in the 6.0.0-RC (release for the XL should follow soon-ish). Closing, please create a new ticket if the issue happens with the 6.0 firmware.

closed 2024-04-13 by CZDanol