prusa3d/Prusa-Firmware-Buddy #3877: MK4 FW 5.1.3   Layer shift - suggested improvement to detect and prevent (could be applied to 6.0.0)

bryn51, 2024-03-31:
**Printer type** -Mk4 Factory Built, 2 weeks old.
**Optional upgrades** - Prusa Enclosure

**Is your suggested improvement related to an existing problem? Please describe.**
  During a large print, a layer shift occurs with no prior symptoms.  The shift occured (in my case) on Y axis.
Belt Tensions are implicated (both X and Y belt tensions read low 50% of the time). (Prusa Belt tuning App).  Filament moisture is also implicated, however the filament roll is within a dryer, which had timer expired (turned off).
Root cause: a raised deposit of greater than 1 layer in height on the uppermost layer occurs, and next time that extruder and bed moves to the same position, extruder or bed movement is impeded, there is a clash of wills, the belt looses the contest, slips, causing a miscalibration. The print continues but prelative extruder position is incorrect.

Many others report random layer shifts on Mk4. Some blame firmware. I do not think its that, but firmware could be enhanced to prevent the issue.

**Describe the expected functionality**
 The issue of layer shifting is common.  The series of events above are well known.  The firmware could assist to detect and prevent the layer shift.  Some suggestions (ideas) as to how this might be done:
1. Perhaps the load cell could detect the lateral force of the Nextruder striking the raised surface of the print, and take action to check its absolute position before carrying on the print.  In any case, dating back to the MK3S+, firmware is able to detect a collision, perhaps this is not working properly in the MK4. The widespread user experience of layer shifting suggests this is the case. Requires verification.
If the firmware suspects a collision has occurred, it could pause the print, allowing the user to check, and elect to abort or carry on the print. The printer could reset its absolute xy position in the process, and compute from prior gcode the position it is supposed to be, so as to resume the print with no visible artefact.

2. If, when moving the extruder quickly from one part of the print to another (with extrusion on hold), the extruder could move in one vector only : first in (say) X direction first, and THEN y direction, (and not at the same time), there will be no clash of wills between the two steppers, and the belt slippage would be less likely to occur.  This could be a firmware user setting, to give users the illusion of having some control.
There are reasons for thinking this might not be enough on its own to prevent the issue, but it might work in some cases.

3. When a collision is detected, automatically (without user intervention) 
[image]
increase Z lift during travel for the duration of the print.  At minimum, one increment of  <layer height>

 

 

[image]

tkircher, 2024-04-01:
The repeatability of the back-EMF homing method is really bad, so it's not likely that a print can recover from a stall. It does seem like the printer should be smart enough to avoid a stall though.

bryn51, 2024-04-01:
I did not mention the stall detection method, although that might be what is used on a mk3x.
The mk4 has the load cell, but I am not certain if it can detect a lateral force applied to the nozzle.
But whatever the case other methods as detailed could be used to avoid a crash and ultimately print failure.
Publicly, prusa downplay the incidence of layer shifting. In 6.0.0 RC notes it says a bug was found and fixed relating to limited incidence on printers fitted with MMU3. Yeah ok. So thats a different issue then. Good that it's fixed but irrelevant to what I am referring to in this report.

closed 2024-07-02 by github-actions[bot]