prusa3d/Prusa-Firmware-Buddy #3878: 6.0.0-RC1 on MK4 - Confirm "STOP" esp. with Touch Screen Active

ericd1949, 2024-04-01:
MK4 Firmware 6.0.0-RC1

The touch screen works a treat, but now the "STOP" button when printing, is **very** active.

In non-touch screen mode, it takes a couple of turns and a click of the wheel to activate STOP, and that is fine, quite a bit of deliberation required, but in touch screen active mode an accidental brush, when maybe the "TUNE", or "PAUSE" button was the target, is final - no recovery possible.

Settings on the KlipperScreen UI, have an option to "Confirm Emergency Stop" and this suits me very well.

Some people, I understand, might want an instant "STOP" button, but I think an optional "Are You Sure?" would be good.

H0Z0, 2024-04-01:
If you press the Stop button in touch mode, printing does not stop immediately, but a confirmation text appears: "Are you sure to stop this printing?" There you can choose Yes or No.

ericd1949, 2024-04-02:
Ahh right, I'm almost always connected with an App talking to Octoprint, and in that case the MK4 doesn't give that option of Yes/No, if you press the "STOP" on the MK4 touch screen - it just stops.

CZDanol, 2024-04-15:
Hey, this is a good idea. Thanks!

Internal ticket number: BFW-5371

CZDanol, 2024-04-15:
Ahh, we already have that :D Should have read the thread more carefully. But I'll add it to the serial printing as well.

CZDanol, 2024-04-15:
Added in the code. Should get in some upcoming release :)

closed 2024-04-15 by CZDanol

ericd1949, 2024-04-19:
Great, accidentally tested it today on 6.0.0, and now it’s “fixed”. Splendid! Thanks.