prusa3d/Prusa-Firmware-Buddy #3879: [FEATURE REQUEST] Network cam integration in dashboard

dl9sec, 2024-04-01:
Hi,

it would be great to have the possibility to integrate a network cam stream into the dashboard of the MK4 PrusaLink (similar to the control window of Octoprint).

E.g. if a checkbox "Integrate webstream in dashboard" is set and a link like "http(s)://mycam.local:8083/stream.mjpg" is given, the stream is shown beside the tempertaure chart on the dashboard.

So the MK4 integrated webserver doesn't need resources for streaming, just the client browser...

Regards, Sten

shledge, 2024-04-09:
I agree, I also think it would be useful for Prusa Connect. I use a Tapo C200 to overlook my MK4, would be cool to integrate it into Prusalink/Prusa Connect

dl9sec, 2024-06-12:
A statement from the developers would be great. Tnx.

Tomcus, 2024-06-12:
This repository is for the buddy firmware only. For features for Prusa Link you have to create an issue in https://github.com/prusa3d/Prusa-Link-Web. For Connect you have to leave feedback (probably - I personally don't know) on the webpage.

Also even though the MK4 doesn't need to add new functionality we need to store the javascript that we serve in the printer and we are running out of space for MINI. So that's something to be considered too.