prusa3d/Prusa-Firmware-Buddy #3881: [BUG] Mk3.5 with Revo 60w fails hotend test on 5.2.2

keyboard-magpie, 2024-04-02:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI, MK4, XL] MK3.5

**Printer firmware version** - [e.g. 4.0.5, ...] 5.2.2

**Original or Custom firmware** - [Original/Custom] Original

**Optional upgrades** - [e.g. Filament Runout Sensor] Revo HF 60W

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint: USB/Prusaconnect

**Describe the bug**
  A clear and concise description of what the bug is.

My 3.5 with revo 60w heater is consistently failing the hotend test, despite trying: parts fan blasting fully, heating up faster with soldering iron, all three possible options.

**How to reproduce**
  Please describe steps to reproduce the behavior.
  
  Trigger hotend test.

**Expected behavior**
  A clear and concise description of what you expected to happen.
  
  Pass hotend test

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

N/a

**Crash dump file**
  Please send the crash dump file to Prusa by emailing it to reports@prusa3d.com. Sharing this file is important and helps us investigate the bug.

N/a

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

**Video**
  Please attach a video. It usually helps to solve the problem.

antimix, 2024-05-05:
Same problem here. I had the 40W on the MK3S+ but when I upgraded to MK3.5 I replaced it with the 60W version.

closed 2024-06-18 by Prusa-Support

protofun, 2026-11-08:
Having a video alongside a bug report can make reproducing issues so much easier.

I've been working on ReleaseVu, a tool that records software interactions and automatically generates clear reproduction steps and bug reports.

I think workflows like this are exactly where tools like these can help.

https://releasevu.com/