prusa3d/Prusa-Firmware-Buddy #3882: [BUG] Touching screen provokes beeping sound, even if no control touched.

anghelos, 2024-04-02:
**Printer type** - MK4

**Printer firmware version** - 6.0.0-RC1

**Describe the bug**
Touching the screen anywhere emits a beep, even if nothing was actually activated. This can be misleading when missing a button, as the sound feedback makes it seem like the button was tapped, when it wasn't.

**How to reproduce**
Activate touch in the settings, then touch the screen anywhere that isn't a button.

**Expected behavior**
  There should be audible feedback only when a clickable element is touched.

CZDanol, 2024-04-13:
Hey, we know about this and are discussing if whether we should change it or not. The behavior is currently actually consistent with how knob behaves - it also beeps always when you click it, no matter if the click actually does something or not.

Internal ticket: BFW-5048

anghelos, 2024-04-17:
I feel the scenarios where pressing the knob doesn't do anything are extremely limited. It also already has proper tactile feedback.
A beep for every interaction, even if nothing is happening, is useless. Its only use would be if you're troubleshooting the touchscreen.

Tupson444, 2024-07-04:
It's easy to miss a button on the touch screen, and the beep can confuse to make you think that you did press the button, only to realize nothing is happening. At least there should be a different sound when tapping on nothing compared to pressing a button, or simply turn off the unnecessary beeping.

berlincount, 2025-08-10:
I just wanted to file the same bug - and I do consider it a bug as well!

- interacting with a control should have some kind of visual and acoustic feedback
- if the interaction was ineffective, this should be distinguishable

I agree that the current behaviour is consistent with the knob, but - maybe it's just a similar bug, then ;)

my preferred solution would be different sounds in case my interaction did something vs. when it doesn't - e.g. maintain the "beep" when something was done and add a "boop" when turning the knob or touching the display was registered, but effectively did nothing ...

thanks for all the great work! I'm very happy my CORE ONE!

3d-gussner, 2025-08-11:
Sorry to tell you that the request has been declined.
For you the touch beep somewhere on the screen may be a bug for others it is an indicator that the touch works.
If you don't want the sound you can disable it.

3d-gussner, 2025-08-11:
@Tupson444 I understand your request and it might be useful to have different sounds, one click on icon another when you miss.