prusa3d/Prusa-Firmware-Buddy #3884: Movement of the E axis causes the filament to be ejected

Mikyner, 2024-04-04:
Printer type - MK4

Printer firmware version - 6.0.0-RC1

Original or Custom firmware - Original

**Describe the bug**
  A clear and concise description of what the bug is.

How to reproduce
Selecting Control > Move axis > Move E causes the filament to be ejected.

CZDanol, 2024-04-13:
Internal ticket: BFW-5367

danopernis, 2024-04-22:
Hi @Mikyner can you be more specific about what the issue is? When the extruder motor moves, filament is either pushed from the nozzle or ejected from the printer, depending on the direction of the motor. This works as intended. I fail to see what the issue is.

Mikyner, 2024-04-22:
If I select the E axis movement menu item, the filament starts to pull out of the nozzle without turning the knob.

Mikyner, 2024-04-22:
It happened to me whenever I wanted to squeeze more filament after I finished printing.

3d-gussner, 2025-11-19:
Please try firmware [6.4.0-RC2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-RC2) and report back.