prusa3d/Prusa-Firmware-Buddy #3891: [BUG] 6.0.0-RC1 Left swipe closes Purge Filament screen

the-ress, 2024-04-06:
**Printer type** - MK4

**Printer firmware version** - 6.0.0-RC1

**Original or Custom firmware** - Original

**Optional upgrades** - Enclosure

**USB drive or USB/Octoprint** - USB flash drive

**Describe the bug**
Left swipe on Purge Filament screen stops the operation. After tapping Purge Filament again, it seems to continue from where it left off.

The same thing happens at the unload screen and possibly elsewhere.

**How to reproduce**
1. Go to Filament menu
2. Tap Purge Filament
3. Wait for the temperature
4. When the filament is actually purging, swipe from right to left in the middle of the screen
5. _Printer shows the Filament menu again_
6. Tap Purge Filament again
7. _The purge continues from the percentage it was before step 4 (unless the extruder temperature dropped)._

**Expected behavior**
IMO, the swipe left shouldn't do anything at these screens. Or at least it should stop the purge cleanly and start anew the next time.

CZDanol, 2024-04-13:
Thank you for your report. I'll look into it.

Internal ticket: BFW-5366

CZDanol, 2024-04-15:
Checked this and it is correct behavior. The purging does not continue where it left of, it always starts the process from the beginning. The purging move itself reports the progress as 70-100, however, but this is okay - the percentage before that is intended for preheating and such and the re-calculating the progress estimate for all the different situations of what will be done during the process and what not would be quite complicated.

closed 2024-04-15 by CZDanol