prusa3d/Prusa-Firmware-Buddy #3894: [ENHANCEMENT] XL Printing with different Nozzles (hight/brand) in one print

RoHeck, 2024-04-08:
**Printer type** - [XL]

**Printer firmware version** - 5.2.2.]

**Original or Custom firmware** - [Original]

**USB drive or USB/Octoprint**
  USB

**Is your suggested improvement related to an existing problem? Please describe.**
I use original Prusa nozzles, as well as Bondtech CHT nozzles and a Rubin nozzle.
The nozzles are not all the same length, which means that the distance between the nozzle and the print bed is different in each case.
The Z-offset between the individual toolheads is only determined during the toolhead calibration, for which the calibration pin must be used and which takes longer than 15 minutes.
If I now use different nozzles in a print (e.g. a CHT for PLA and a normal one for TPU), then there is a Z offset between the two nozzles that is not compensated for and leads to adhesion problems. 
The only way at the moment is to carry out a complete new toolhead calibration. But that takes far too long, especially as I change my nozzles every 2-3 prints and adapt them to my printing requirements.

**Describe the expected functionality**
In my opinion, one solution would be to carry out a nozzle cleaning and subsequent Z-homing for each toolhead used before a print and thus redetermine the Z-offset between the tools (the homing must of course be carried out at the same point with each tool in order to compensate for variations in the bed).

mtaumike, 2024-04-11:
I do not agree - I would just go through the calibration though not with the current 6.0.0 code cause it is br0ke.

Why?

These nozzles (any of them) are not necessarily the same exact diameter (not the hole), length, and could actually be a little oval vs perfectly circular. Going through the calibration will make sure that as far as the printer is concerned, the squirting hole is aligned on all 3 axis.

smurfix, 2024-04-19:
Huh. IMHO if you have a nozzle that's off-center in X or Y then you have far worse problems than can be fixed with Prusa's 3-axis calibration.

Z-homing each tool would fix this. Don't even need to do it before every print – just heat up the tool you just changed the nozzle on, plus some other tool that you left alone, and do the z-homing plus nozzle cleaning thing.

RoHeck, 2024-04-19:
@smurfix you are right. An option in the menu that allows one (or more) toolheads to be recalibrated in relation to another toolhead in the Z direction (preferably with automatic prior heating and NozzleCleaning) would simply solve the problem.
This would then also not affect the start time of a print if no nozzle was changed.

mstandif, 2024-07-06:
I agree this should be a feature. 3-axis calibration is overkill if the only thing being switched is one of the nozzles.

Either an additional Z-axis calibration menu item should be added to quickly redo the z-offsets between each toolhead, or add a new toggle to enable checks before each print. The printer already grabs every toolhead that will be used for the print, so it wouldn't add much time to probe too.

closed 2026-10-05 by github-actions[bot]