prusa3d/Prusa-Firmware-Buddy #3896: [BUG] heating/cooling to 170C during the MBL delays the start of printing by minutes! (dup of issue 3845)

woecj, 2024-04-08:
This annoying behavior was alrady reported in issue 3845, but seems not to be solved.

**Printer type** - [MK3.5]

**Printer firmware version** - [5.5.2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - []

**USB drive or USB-Drive
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
 The additional step of heating/cooling to 170C during the MBL on the MK3.5 delays the start of printing by minutes!

**How to reproduce**
 Everytime you Start the print.

**Expected behavior**
  Direkt heating nozzle/bed to intended temperature (behavior like the MK3S+ before)

Please check the firmware!

Tupson444, 2024-04-08:
Just change the start code in PrusaSlicer (Printer Settings > Custom G-code > Start G-Code). Search for command `M109 R` or `M109 T0 R` (this sets the temperature and commands it to wait until reached). You can find more info about Gcode here: https://marlinfw.org/docs/gcode/M109.html

I guess if the printer profile for MK3.5 in PrusaSlicer is not optimal, you should open a request here https://github.com/prusa3d/PrusaSlicer-settings (if I'm not wrong).

woecj, 2024-04-08:
Ok, but as in issue 3845 mentioned, this behavior appears also by doing a first sheet calibration - which is issued by the printer w/o having commands from the slicer  - so it seems to be in the firmware!

Tupson444, 2024-04-08:
I'm just assuming (I have MK4, not MK3.5), but even if the firmware has these commands for first sheet calibration, changing the start code in the slicer should change the behavior of the printer, although I may be wrong (someone else has to confirm this).

H0Z0, 2024-04-09:
> Ok, but as in issue 3845 mentioned, this behavior appears also by doing a first sheet calibration - which is issued by the printer w/o having commands from the slicer - so it seems to be in the firmware!

First sheet calibration and pre-printing warm-up are two different things. The heating before printing can be changed in G-code if someone wants it. Although the MK3.5 does not use a loadcell, the lower temperature helps to prevent unnecessary leakage, which is a problem in any case, even when using SPINDA.
The first sheet calibration routine is indeed performed by FW. However, there is no need for a higher temperature during the measurements, but it also prevents leakage there during calibration. So this is not a bug, but rather a modified operation, which is more of an advantage.

woecj, 2024-04-09:
I understand, that it is a good idea to do the SPINDA measurements at lower temperature, but I do not understand, why the nozzle is heated up to 230° before starting the SPINDA measurements.
The procedure on printing is as follows: heatup to print temperature (230° for PLA), then wait to cool down to 170°, then make the PINDA measurements, then heat up to print temperature and start printing.
So the first heatup is senseless, even it produces leakage!

H0Z0, 2024-04-09:
The official starting G-code for MK3.5 has only this before bed leveling:
M104 T0 S170 ; set extruder temp for bed leveling
M109 T0 R170 ; wait for temp
There is this line only after bed leveling:
M104 S{first_layer_temperature[0]}
So, if you have include this lines in this order, shouldn't have to heat the hotend up to the printing temperature and cool it down from there when you start printing. I think you should check it, maybe you have a print temp heating line before bed leveling.

woecj, 2024-04-09:
This is what I can find as start code and what leads to the sequence I described:

`M862.3 P "[printer_model]" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M115 U3.13.2 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 X{first_layer_print_min[0]} Y{first_layer_print_min[1]} W{(first_layer_print_max[0]) - (first_layer_print_min[0])} H{(first_layer_print_max[1]) - (first_layer_print_min[1])} ; mesh bed levelling
{if filament_settings_id[initial_tool]=~/.*Prusament PA11.*/}
G1 Z0.3 F720
G1 Y-3 F1000 ; go outside print area
G92 E0
G1 X60 E9 F1000 ; intro line
G1 X100 E9 F1000 ; intro line
{else}
G1 Z0.2 F720
G1 Y-3 F1000 ; go outside print area
G92 E0
G1 X60 E9 F1000 ; intro line
G1 X100 E12.5 F1000 ; intro line
{endif}
G92 E0
M221 S{if layer_height<0.075}100{else}95{endif}

; Don't change E values below. Excessive value can damage the printer.
{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3).*/}M907 E430 ; set extruder motor current{endif}
{if print_settings_id=~/.*(SPEED @MK3|DRAFT @MK3).*/}M907 E538 ; set extruder motor current{endif}`

H0Z0, 2024-04-09:
Then that's the problem. This is not MK3.5 Start G-code, but MK3. In the slicer, change the printer type to MK3.5. If it is not already in your list, add it in the configuration wizard.

lezetuser, 2024-04-10:
The mk3/ 3.5 PINDA has the thermal compensation unlike MINI. It can do MBL at printing temperature for better calibration. MINI  does not have  and MK4  does not need, so they heat the nozzle only to the non leaking  temperature to be almost ready for printing

rtyr, 2024-04-11:
It looks this happens when running MK3 gcode on MK3.5 printer, likely unwanted FW behavior, forcing 170C for MBL even when it is not included in the start gcode.

CZDanol, 2024-04-13:
Internal ticked number: BFW-5365

Anubis1971, 2024-04-23:
Is there way to get around this issue for 3.5?

Anubis1971, 2024-04-26:
I use the correct 3.5 profile for my printer but it still heats or cools to 170C, MBL, then heats to required temp before every print start.  It’s quite annoying, when needing to start and stop prints for testing or errors/failures of some kind.  Ideally, I would like to: initiate print, MBL immediately, heat to correct temp, start print.

H0Z0, 2024-04-26:
There is no line in starting G-code for MK3.5 profile that heats the nozzle to print temperature before the bed level. It heats only the nozzle to 170°C and heats it to the set printing temperature only after the bed level. What is included in your starting G-code?

timur-tabi, 2024-07-03:
Not stale, just ignored by Prusa.

ghouscht, 2024-09-03:
This is the start G-code I got in PrusaSlicer (v2.8.0) for my MK3.5:
```gcode
M17 ; enable steppers
M862.1 P[nozzle_diameter] ; nozzle diameter check
M862.3 P "MK3.5" ; printer model check
M862.5 P2 ; g-code level check
M862.6 P"Input shaper" ; FW feature check
M115 U6.0.4+14924

M555 X{(min(print_bed_max[0], first_layer_print_min[0] + 32) - 32)} Y{(max(0, first_layer_print_min[1]) - 4)} W{((min(print_bed_max[0], max(first_layer_print_min[0] + 32, first_layer_print_max[0])))) - ((min(print_bed_max[0], first_layer_print_min[0] + 32) - 32))} H{((first_layer_print_max[1])) - ((max(0, first_layer_print_min[1]) - 4))}

G90 ; use absolute coordinates
M83 ; extruder relative mode

M140 S[first_layer_bed_temperature] ; set bed temp
M104 T0 S170 ; set extruder temp for bed leveling
M109 T0 R170 ; wait for temp
M190 S[first_layer_bed_temperature] ; wait for bed temp

G28 ; home all

G29 P1 ; invalidate mbl & probe print area
G29 P1 X23 Y5 W80 H20 C ; probe near purge place
G29 P3.2 ; interpolate mbl probes
G29 P3.13 ; extrapolate mbl outside probe area
G29 A ; activate mbl

; prepare for purge
M104 S{first_layer_temperature[0]}
G0 X0 Y-4 Z15 F4800 ; move away and ready for the purge
M109 S{first_layer_temperature[0]}

; Extrude purge line

G92 E0 ; reset extruder position
G0 E7 X15 Z0.2 F500 ; purge
G0 X25 E4 F500 ; purge
G0 X35 E4 F650 ; purge
G0 X45 E4 F800 ; purge
G0 X{45 + 3} Z0.05 F8000 ; wipe, move close to the bed
G0 X{45 + 3 * 2} Z0.2 F8000 ; wipe, move quickly away from the bed

G92 E0
M221 S100 ; reset flow to 100%
```

In theory you could simply replace these two lines in the start G-code
```gcode
M104 T0 S170 ; set extruder temp for bed leveling
M109 T0 R170 ; wait for temp
```

with this
```gcode
M104 S{first_layer_temperature[0]}
M109 S{first_layer_temperature[0]}
```

to heat directly to the first layer temp.