prusa3d/Prusa-Firmware-Buddy #3897: [BUG] MK4 refuses to nozzle clean or level unless I reset printer on 6.0 RC

shledge, 2024-04-09:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 6.0 RC

**Original or Custom firmware** - Original

**Optional upgrades** - none

**USB drive or USB/Octoprint**
USB Drive

**Describe the bug**
Nozzle cleaning or level probing refuses to work unless I reset printer. When I complete or stop a print after said reset, it once again refuses to nozzle clean or level until I reset it again. Nozzle is spotless and even tried changing probe temperature to 150c in start gcode to prevent oozing, which still does not fix the issue.

Downgrading back to 5.13 fixed the issue for me, were it works every time as expected.

**How to reproduce**
Reset printer, then start a print. Either stop it or let it finish, then immediately start another print. Always fails and only will work again until you reset printer again.

**Expected behavior**
Nozzle cleaning and levelling works as expected.

TheYang, 2024-04-15:
Same Behaviour on XL

shledge, 2024-04-16:
Updated to 6.0 RC2, still seems to work okay now like it did before on 5.1.3. Not really sure why it happened as there was no code changes related to this issue.

CZDanol, 2024-04-16:
There's been a bunch of fixes between RC1 and RC2, most of them didn't get to the release notes because they were not significant enough and too complicated to even explain.

TheYang, 2024-04-17:
Seems to not be improved on XL, just needed to reset again on 6.0.0-RC3

danopernis, 2024-04-18:
@shledge @TheYang Thanks for the bug report. I see several similar issues open already, so let's keep all the relevant info in one issue, which will be https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3818 as it was the first as far as I can tell.

closed 2024-04-18 by danopernis