prusa3d/Prusa-Firmware-Buddy #3898: [FEATURE REQUEST] Ability to skip nozzle cleaning feature

shledge, 2024-04-09:
**Printer type** - MK4

**Is your feature request related to an existing problem? Please describe.**
Allow user the ability to skip nozzle cleaning, especially for problematic filaments such as TPU or PC/PCCF, were nozzle cleaning always fails.

**Describe the expected functionality**
Give a prompt on screen to allow user to skip, either by tapping on a button on screen, or clicking the encoder.

danopernis, 2024-04-11:
Nozzle cleaning is part of the generated gcode, so this should be opened in [PrusaSlicer. ](https://github.com/prusa3d/PrusaSlicer)

closed 2024-04-11 by danopernis

shledge, 2024-04-11:
> Nozzle cleaning is part of the generated gcode, so this should be opened in [PrusaSlicer. ](https://github.com/prusa3d/PrusaSlicer)

I'm aware of this (in fact I have it as an if/else statement in start gcode so it currently doesn't happen if I use TPU or PCCF), but I'd rather let it attempt if needed.