prusa3d/Prusa-Firmware-Buddy #3900: [ENHANCEMENT] Increase maximum steps/mm per extruder

ClaGre, 2024-04-09:
After upgrading to MK3.5 I decided to upgrade extrusion system with Sherpa.
Designed, installed, I wanted to vary the steps/mm and by accessing the hidden Hardware menu I increased the value which however has a maximum of 1000 (in my case I should enter a value just under 1300); I am also thinking of those who perhaps in the future will want to install an orbiter system and perhaps the value will have to be even greater.
I ask if it is possible to increase this value in the next release by 3000.

**Printer type** - [MK3.5]

I tried to enter the command:
M92 E1261
E500
via Pronterface but it doesn't remain in memory.
Thanks

ClaGre, 2024-04-16:
Update: I set the correct value at G-code level on M92 after which I encountered the problem in unloading the filament and I added 3mm to the displacements of the filament in the G-code end.
Having solved this I found a new problem; when I ask MMU3 to load the filament into the nozzle he takes it to the filament sensor then the extruder motor follows the value set by the firmware (1000) and fails to engage correctly.
It is absolutely necessary to have the possibility to insert a step/mm value from the firmware for the extruder motor greater than 1000 and also to set the PTFE tube length and any distance between the PTFE tube and the knurled wheel.
Everything else works correctly, providing dimensionally correct prints.

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/102923571/d9b5ab1a-c025-461b-8309-c606a7eb41af

[image]
[image]
[image]
[image]
[image]
[image]
[image]

ClaGre, 2024-04-18:
@danopernis Good morning, can you confirm that the steps/mm value for the extruder has a maximum value of 1000 and this value is NOT (currently) modifiable even by sending the command M92 Exxxx (where xxxx is a value greater than 1000) and then M500 for save the setting?

When I tried to enter the commands described above and then doing M503 it returns the value I entered (e.g. 1330) but then if I send the M501 command from the EEPROM it reads 1000.

Therefore the value is NOT saved in memory.

danopernis, 2024-04-18:
Better to be answered by @DRracer

ClaGre, 2024-04-25:
> @danopernis @DRracer good morning, can you confirm that the steps/mm value for the extruder has a maximum value of 1000 and this value is NOT (currently) modifiable even by sending the command M92 Exxxx (where xxxx is a value greater than 1000) and then M500 for save the setting?
> 
> When I tried to enter the commands described above and then doing M503 it returns the value I entered (e.g. 1330) but then if I send the M501 command from the EEPROM it reads 1000.
> 
> Therefore the value is NOT saved in memory.

@DRracer can you confirm my supposition or can you suggest a solution? Thanks

ClaGre, 2024-05-06:
@danopernis I'm sorry to mention u again but @DRracer don't answer me 🤷🏻‍♂️
I've read firmware files and maybe found the right code:

File: src/gui/menu_vars.h

if (PRINTER_IS_PRUSA_MK4 || PRINTER_IS_PRUSA_MK3_5 || PRINTER_IS_PRUSA_XL || PRINTER_IS_PRUSA_iX)
    constexpr static std::array<int, RANGE_SZ> feedrate_range = { 50, 1000, 1 };
#else
    constexpr static std::array<int, RANGE_SZ> feedrate_range = { 10, 255, 1 };
#endif
    constexpr static std::array<int, MenuVars::RANGE_SZ> microstep_exponential_range = { 1, 256, 2 }; // 2^0 - 2^8 .. 1, 2, 4, .. , 128, 256
    constexpr static std::array<int, MenuVars::RANGE_SZ> microstep_exponential_range_with_0 = { 0, 256, 2 }; // 0 + 2^0 - 2^8 .. 0, 1, 2, 4, .. , 128, 256
    constexpr static std::array<int, MenuVars::RANGE_SZ> axis_rms_currents_range = { 0, 800, 1 };
    constexpr static std::array<int, MenuVars::RANGE_SZ> steps_per_unit_range = { 1, 1000, 1 }; // small range, experimental feature

Into the last line there is set steps_per_unit_range with max value of 1000 in I ask to increase at 1500 to include more reduction gear into extruder mods.

There is possibility for the developers to include the modify into next releases?
Otherwise we have to find others solutions.
Thanks again

closed 2024-07-15 by github-actions[bot]