prusa3d/Prusa-Firmware-Buddy #3901: [QUESTION] Enable translations

xtomasnemec, 2024-04-11:
Please, before you create a new question, please make sure you searched in open and closed issues and couldn't find anything that matches.

How can i enable translations while compiling MINI firmware?
[image]
There aren't any flags to do so.

danopernis, 2024-04-11:
There is a cmake option for that. Or you can use a preset. How exactly are you trying to build the firmware?

https://github.com/search?q=repo%3Aprusa3d%2FPrusa-Firmware-Buddy%20TRANSLATIONS_LIST&type=code

xtomasnemec, 2024-04-11:
> There is a cmake option for that. Or you can use a preset. How exactly are you trying to build the firmware?
Not in debug mode

xtomasnemec, 2024-04-11:
nevermind i used --help

closed 2024-04-11 by xtomasnemec