prusa3d/Prusa-Firmware-Buddy #3902: [BUG] Firmware gets confused when auto-entering Paused state, disables hot ends

smurfix, 2024-04-11:
**Printer type** - XL, multi-head but most likely doesn't matter

**Printer firmware version** - 5.1.2

**Original or Custom firmware** - Original

**Optional upgrades** - none

**USB drive or USB/Octoprint** - Octoprint

**Describe the bug**
  When the firmware enters Paused state due to failed probing, it subsequently gets confused and disables the hot end heaters.

  The display shows "tune / resume / cancel", with a Resume that's not selectable.

  It does not seem possible to return the printer to a working state via G-Codes.

**How to reproduce**
  See the [annotated serial log](https://github.com/OctoPrint/OctoPrint/issues/4977#issuecomment-2049177585) in the Octoprint issue.

Note the line `// Another probe, firmware sets target of T1 to 0!` !!

  The log shows that one of M302, M84 or G29 manages to turn off the hot end as a side effect when the printer is in this state.

**Expected behavior**
  The printer gets out of Paused state with an M602, M603, or simply by starting a new print.

  It doesn't turn off heaters mid-job without reporting an error.

  Also, if the printer thinks that its hot ends need to be disabled it should return an error message when Octoprint tries to set the temperature. It should not just continue to "print".
  
**G-code**
  See the [serial log](https://github.com/OctoPrint/OctoPrint/issues/4977#issuecomment-2049177585).

smurfix, 2024-08-12:
Any thoughts?

smurfix, 2024-10-14:
problem still exists AFAIK

svenstaro, 2025-01-18:
This sounds related to https://github.com/prusa3d/Prusa-Firmware/issues/4052 even though it's a completely different firmware.

CZDanol, 2025-01-21:
Hello, the probing fail will be handled differently in the future firmware (after 6.3), so this issue should get resolved then.

closed 2025-03-31 by github-actions[bot]