prusa3d/Prusa-Firmware-Buddy #3905: Ability to skip nozzle cleaning during print

shledge, 2024-04-12:
Putting this up again as I felt there was no good reason to close this - see quote at end. The current solution with start gcode is inelegant and not really obvious to new users.

Basically have the ability to skip nozzle cleaning during the start of a print, especially for more problematic filaments like PCCF were nozzle cleaning tends to fail.

Could literally be allowing the user to select "skip" *when* nozzle cleaning fails.

              > Nozzle cleaning is part of the generated gcode, so this should be opened in [PrusaSlicer. ](https://github.com/prusa3d/PrusaSlicer)

I'm aware of this (in fact I have it as an if/else statement in start gcode so it currently doesn't happen if I use TPU or PCCF), but I'd rather let it attempt if needed. I don't think this should be something that needs to be closed IMO.

_Originally posted by @shledge in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3898#issuecomment-2049424151_

CZDanol, 2024-04-13:
Hello, nozzle cleaning is needed for the MBL to work correctly. If you do bed leveling with some debris on the nozzle, the results will get skewed.

shledge, 2024-04-16:
The problem is that it fails either way on materials like TPU or PCCF, unless I reduce the probing temperature - which in itself will also skew the levelling.

At least allowing me to continue the print, even if nozzle cleaning failed, means the nozzle is at least in a "good enough" state to level because of said attempt in the first place.

shledge, 2024-04-16:
> Hello, nozzle cleaning is needed for the MBL to work correctly. If you do bed leveling with some debris on the nozzle, the results will get skewed.

The problem is that it fails either way on materials like TPU or PCCF, unless I reduce the probing temperature - which in itself will also skew the levelling.

At least allowing me to continue the print, even if nozzle cleaning failed, means the nozzle is at least in a "good enough" state to level because of said attempt in the first place.

CZDanol, 2024-04-16:
I guess that is valid. I'll try to discuss it in the office.

CZDanol, 2024-04-17:
Aight, I've created an internal ticket BFW-5394 for this. No promises on when it will be done or if it will be done at all though.