prusa3d/Prusa-Firmware-Buddy #3906: [BUG] Firmware 6.0.0-RC2 for MK3.5 not compatible with MMU3 latest firmware.

peterjhl, 2024-04-12:
**Printer type** - [MK3.5]

**Printer firmware version** - [6.0.0-RC2]

**Original or Custom firmware** - [Original]

**Describe the bug**
Firmware 6.0.0RC2 for MK3.5 not compatible with MMU3 latest FW3.0.2+878 firmware.

[image]

cozmo14047, 2024-04-12:
Hi there, yes i had this issue turns out you have to flash the mk4 mmu firmware which is the 896 version here, i beleive its an error in the documentation. this way works and is working well with mine https://github.com/prusa3d/Prusa-Firmware-Buddy/releases

Patrick-Schmitt, 2024-04-12:
Hi same situation with my MK3.5 and MMU3. I simply flashed the MMU3 firmware which is "only" for MK4 and MK3.9 but it worked without any issues. Currently printing something with MMU so no issues there.
Probably just a bad documentation.

cozmo14047, 2024-04-12:
> Hi same situation with my MK3.5 and MMU3. I simply flashed the MMU3 firmware which is "only" for MK4 and MK3.9 but it worked without any issues. Currently printing something with MMU so no issues there. Probably just a bad documentation.

yeah thats what i thought, mine is working fine except the timeout period for hitting the finda is far to short now

Patrick-Schmitt, 2024-04-12:
> yeah thats what i thought, mine is working fine except the timeout period for hitting the finda is far to short now

Really? Ohoh... will have to monitor that. Will report if I have the same issue. (of course in the correct issue topic)

CZDanol, 2024-04-13:
Hey, I've reported it to the docs guys, where does it say/not say the correct FW version specifically?

Patrick-Schmitt, 2024-04-13:
> Hey, I've reported it to the docs guys, where does it say/not say the correct FW version specifically?

In the release notes of the latest RC candidate for the MK3.5, MK3.9 and MK4 firmware (6.0.0-RC2) at the chapter "MMU3 unit firmware auto-update".
There is the following text:

> MMU firmware compatibility (files are included below):
> FW3.0.3+896.hex - compatible with MK4, MK3.9
> FW3.0.2+878.hex - compatible with MK3.5

And that is causing the confusion. Yes, it is correct that FW3.0.2+878.hex will only work with MK3.5 and MK3 BUT for the new 6.0.0-RC2 to work you will have to update the MMU firmware to "FW3.0.3+896.hex" which does work completely fine with MK3.5.

My guess is that this was carried over from the previous release notes of firmware 6.0.0-RC1. This release was really only for MK3.9 and MK4. Here is the part from the release notes of RC1 which is probably why this was carried over the RC2 as well:

> Please note, that unit's firmware 3.0.3 must be paired with the printer's firmware 6.0.0-RC1. No other version combination for MK4 is possible.
> Note: The MMU3 firmware 3.0.3 can't be used with MK3.5 and MK3S+.

I hope my assumptions are correct there - please ask me anything if you need more guidance. Happy to help!

BTW: Currently printing a long high amount of filament switches with my MK3.5 + MMU3 running on firmware 6.0.0-RC2 - it works like a dream!

cozmo14047, 2024-04-13:
Yes except for the intial load where there isint enouhh time left for it to reach finda

peterjhl, 2024-04-13:
> Hey, I've reported it to the docs guys, where does it say/not say the correct FW version specifically?

[image]

CZDanol, 2024-04-13:
> Yes except for the intial load where there isint enouhh time left for it to reach finda

I don't get it.

panEarth, 2024-04-13:
I can confirm that FW3.0.3+896.hex (compatible with MK4, MK3.9) work great on my MK3.5 
[image]

closed 2024-04-17 by CZDanol