prusa3d/Prusa-Firmware-Buddy #3907: mk3.5 mmu3 not enough time to hit finda

cozmo14047, 2024-04-12:
hi there, so for the mk3.5 mmu3 new firmware just released 7 hrs ago it doesent give enough time to hit the finda. i have to push it to the finda, letting the mmu feed it is too slow and timing out. please make the timeout period longer so it has more time to hit the finda thanks.

Patrick-Schmitt, 2024-04-13:
Hi answering you here at the correct issue tracker:

I am sorry I was unable to verify that. Works well here. Have you switched back to the older 5.2.2 firmware for verification?

cozmo14047, 2024-04-14:
> Hi answering you here at the correct issue tracker:
> 
> I am sorry I was unable to verify that. Works well here. Have you switched back to the older 5.2.2 firmware for verification?

Hi, yes the old version works fine.

CZDanol, 2024-04-15:
During which operation does this happen?

cozmo14047, 2024-04-15:
> During which operation does this happen?

if you start a print with no fillament preloaded to nozzle.

CZDanol, 2024-04-16:
If you

DRracer, 2024-04-17:
The default distance limit in the FW is twice the distance from the pulley into FINDA. 
So in you case - either the filament is not grabbed by the gears (which means it is not preloaded correctly) or it gets stuck on its way out (which means you should check the HW of your unit).

CZDanol, 2024-04-17:
Just to be clear - what sensor are you referring to by saying "finda" @cozmo14047 ? The one on the MMU, or the one at the extrude?

cozmo14047, 2024-04-17:
> Just to be clear - what sensor are you referring to by saying "finda" @cozmo14047 ? The one on the MMU, or the one at the extruder?

ok it seems fixed now with finail release, havent tested it fully though. the finda is the mmu one it only happend if you start a print with no fillament preloaded to nozzle.

DRracer, 2024-04-18:
It has definitely **not** "been fixed in the final release", in fact we haven't touched this procedure at all. I'm still curious if we are talking about the same. Do you have a video of the original issue you reported?

cozmo14047, 2024-04-18:
> It has definitely **not** "_been fixed in the final release_", in fact we haven't touched this procedure at all. I'm still curious if we are talking about the same. Do you have a video of the original issue you reported?

no but tommorow i can downgrade firmware and get a video, it does seem to be ok now im not sure if it was a fluke or what but ti definitly happend a few times on the other firmware

CZDanol, 2024-04-18:
If this starts happening again, the video would be great :)

cozmo14047, 2024-04-18:
I see that you havent changed this code but deffinitly carnt reproduce it on the new stable release, ive got a few long prints to do then ill downgrade to get a video if u want

closed 2024-06-27 by danopernis