prusa3d/Prusa-Firmware-Buddy #3908: [BUG] After updating MINI to 6.0.0-RC2 WPA2 does not work

Mathos96, 2024-04-13:
Hi, I updated my MINI to the newest firmware 6.0.0-RC2 and from that moment Wi-Fi can just connect to my wireless router. After some tinkering, I found out it was a WPA security issue. When I downgraded from WPA3, WPA2 to WPA2, WPA it worked. I tried upgrading back to WPA3, WPA2 and the problem was back.

DRracer, 2024-04-16:
Thank you for your report, but WPA2 has been working reliably for us and our beta testers so far. We don't think only the new firmware is to blame, more likely your problem will be a combination of factors.

danopernis, 2024-04-16:
See also https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3864

Mathos96, 2024-04-16:
My router is up-to-date, and I have already tried refreshing my wifi credentials. Yes, my router is running WPA3/WPA2 Transition mode. Now I have tried WPA2 only and it works!

Thanks for helping @dzuelke 😄.

closed 2024-04-16 by Mathos96

danopernis, 2024-04-17:
Thank you folks for taking time to report and debug this issue. Let's move further discussions to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3864