prusa3d/Prusa-Firmware-Buddy #3909: [BUG] MMU3 Finda state not translated in Footer on MK4

unclej84, 2024-04-13:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4 with MMU3

**Printer firmware version** - 6.0.0-RC2

**Original or Custom firmware** - Original

**Optional upgrades** - MMU3

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
The nextruder filament sensor shows german translated "AN"/"AUS" but MMU3 Finda shows "On"/"Off"

**How to reproduce**
Simply add Finda state to footer

**Expected behavior**
State to be also translated for Finda

CZDanol, 2024-04-15:
Internal ticket: [BFW-5373](https://dev.prusa3d.com/browse/BFW-5373)

CZDanol, 2024-04-15:
Fixed, should get into a release eventually.

closed 2024-04-15 by CZDanol