prusa3d/Prusa-Firmware-Buddy #3911: [BUG] MK3.5 unable to complete XY selftest with default range of XY_STALL_SENSITIVITY and aftermarket motors

m-kozlowski, 2024-04-13:
**Printer type** - MK3.5

**Printer firmware version** - 5.2.0 - 6.0.0-RC2

**Original or Custom firmware** - Original

**Optional upgrades** 
All motors except E are part of Triangle Lab's kit dedicated for prusa i3. Exact specifications unknown.

**USB drive or USB/Octoprint**
  NA

**Describe the bug**
  During self test of both X and Y axis Stallguard improperly detected crash on every move in home direction causing carriage/bed to shift slightly to the oposite end of the axis, eventually hitting a frame.
Adjusting belt tension did not help.
Rods are in good condition, parallel and the carriage/bed when pushed manually is moving smoothly, without excessive resistance.
Tried to lower sensitivity with M914 gcode, but despite replying "ok", firmware ignored any changes.
Increasing range of XY_STALL_SENSITIVITY from -7..-2 to -2..5 allowed self test to pass without issues, with 
`calibration offset: 0
echo:Probe classified as perfect.`
on both axis.

**Expected behavior**
  I'm aware that supporting aftermarket parts might not be your top priority, but some way to work around this issue without necessity of building and flashing own firmware would be nice;)

ClaGre, 2024-04-14:
Have you try to increase a little bit motor current?
G-code M906
Find spec here: https://help.prusa3d.com/article/buddy-firmware-specific-g-code-commands_633112

m-kozlowski, 2024-04-14:
Oh, sorry, forgot to mention that:)
I tried increasing it first to 450mA and then to 550mA. It didn't help.

ClaGre, 2024-04-14:
> Oh, sorry, forgot to mention that:)
> 
> I tried increasing it first to 450mA and then to 550mA. It didn't help.

No problem ;)
"Normally" this motor accept nominal current of 1A then you can try to increase at 800/900mA because this is the maximum current your motor can receive but if it need only 500mA (for example in the center of axis where normally the motion is more free) it use only the current needed.

m-kozlowski, 2024-04-14:
At 900mA same error occurs, but much louder:)

closed 2024-07-10 by github-actions[bot]