prusa3d/Prusa-Firmware-Buddy #3913: web Link vs slicer corrupted transmission bug

HonzaJaros, 2024-04-15:
Hello,

yesterday I experienced very strange issue and I believe it is a bug.

my friend send a gcode, 50MB, through the link website through VPN and it was very slow. it shows percentage and when it reached 100% it was sitting there 10-15 and nothing was happening.

so he decided sent the gcode through the slicer (also through the same VPN) after  15 minutes the printer started to print.

after few minutes of first layer I heard wrong sound and found out the nozzle is digging through the smooth sheet, so I stopped it. and went for support.

they observed gcode and did not find anything strange.

so there is my only viable theory, why this happened. 

the slicer somehow haven't been notified that there is already file transaction through the link web and accepted the slicer. meanwhile printing. the link finally got all the data and initiated print as well. corrupted the USB and went Z below 0.

after stoping. the Z was uncalibrated, USB was corrupted and must have been formatted.

no other evidence what happened. support didn't find anything wrong.

VH-fixer, 2024-04-29:
Since I went to FW6.0.0 I have transfered files via wifi and found a few problems.  In one case the wifi load repeated a block of the G code and an area of the print was repeated at the same Z.   This was a first layer and very noticeable.   I deleted and reloaded the G code and everything works correctly. 
Other wifi issues with 6.0.0 I have seen are in other bug reports.   I think the WIFI in fw 6.0.0 needs major quality work.

LuckyResistor, 2024-04-30:
I had similar issues, but my PrusaXL printer is connected via Ethernet.

- In one case, the printer skipped some gcode commands while printing an infix. This resulted in a strange diagonal movement, dragging filament across the part. One line of the infill was missing after this.
- On a second case, the printer repeated a section of infill, printing over the same area twice - what resulted in a huge mess with filament blobs that eventually caused the printer head crashing into a blob, what shifted print. I had to cancel it.

- Investigating the gcode from the SD card and the downloaded versions from Prusa connect didn't had these anomalies.

Is the gcode that is streamed verified with checksums in the firmware? Or is it accepted and printed unverified?

closed 2024-07-15 by github-actions[bot]