prusa3d/Prusa-Firmware-Buddy #3915: [ENHANCEMENT] Add adjustable-temperature filament type

smurfix, 2024-04-15:
**Printer type** - all

**Is your suggested improvement related to an existing problem? Please describe.**
Some filaments have narrow temperature ranges which the printer's hard-coded list of filament types doesn't cover.

For instance, some filaments for lost-form casting should be extruded at ~155°C. Loading them at 210° risks melting too much of it ínside the extruder; when printing with them I sometimes get a clogged nozzle.

**Describe the expected functionality**
Add a "custom" filament type (or two) whose load/unload temperature can be changed somewhere in Settings, and/or with some G-code commands.

murk-sy, 2024-04-30:
Related to PR https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3657
Somewhat duplicate of issues mentioned in PR which request addition of specific filament types

CZDanol, 2024-06-01:
Hey, I will try to push for adding support for custom material types. Unifying the related issues under the internal ticket BFW-5513

CZDanol, 2024-09-11:
Hello, we've released a 6.2.0-alpha1 firmware with user material types support: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-alpha1