prusa3d/Prusa-Firmware-Buddy #3916: [ENHANCEMENT] Guided cold pull on LCD screen

Claghorn, 2024-04-15:
My printer is MK4, but this would probably be handy for other printer models as well.

Doing a cold pull to clean the nozzle means looking up the instructions online, then manually trying to follow them while keeping track of temperature displayed on screen and finding your place in the instructions again after looking away, and when you are finally done, the firmware is confused about whether you have filament installed or not.

If the firmware had a cold pull option you could pick, it could give you onscreen instructions and keep track of the proper temperatures for you, telling you when to pull, etc. and it would know there wasn't any filament installed when done.

mody, 2024-04-16:
Great idea. And even better, it's already implemented! See https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0-RC1

closed 2024-04-16 by mody

Claghorn, 2024-04-16:
Obviously I didn't search enough before adding this :-). Good to know!