prusa3d/Prusa-Firmware-Buddy #3921: [FEATURE REQUEST] "Shipping mode" the home and park the print head in the correct position for shipping it while putting fans on maximum

RichiH, 2024-04-17:
**Printer type** - Mk4
**Optional upgrades** - n/a

I would like the Mk4, and optionally the Mini, to have a "park into shipping position" feature. It would
1. Put fans on 100% to cool down the printhead
2. Home the head
3. Move head into position so it fits the foam cutouts provided by Prusa
4. Display a "You can now pack up your printer. Don't forget to take a photo before closing the box"

My main use would be for Grafana Labs: We have one Mk4 in the EU and the US each to showcase our data visualization software at conference booths and events. This means shipping the printers around a lot and not everyone is familiar with the printer.

As such, I see three main use cases:
* For Prusa Research themselves, to make packing up assembled printers more convenient and quicker for Prusa staff
* For Prusa users, when shipping the printer around
* For both Prusa Research & Prusa users when shipping back the printer to Prusa for repairs

danopernis, 2024-04-17:
This seems like a very niche usecase to put into the firmware. I suggest crafting a custom gcode file with just the required operations and puting that on the USB. You can "print" such a file without any problems.

RichiH, 2024-04-17:
While it could be handled in GCode, the overhead for use cases 1 & 3 would be significant and as such unlikely.

The cost in firmware is neglible and putting it near "factory reset" in the UI would make sense.

danopernis, 2024-04-17:
> The cost in firmware is neglible

Tell that to the STM32F4 chip with 2MB of FLASH and 256kB of RAM :smile: and MINI has even less than that. FWIW our printing farm uses exactly this solution for exactly this reason. It also allows customization without hardcoding everything to the firmware.

CZDanol, 2024-04-20:
Closing this issue.
* This feature can be easily done by creating a custom gcode.
* The gcode would be adjustable, so more versatile than hard-coded firmware function.
* Making it a firmware function would require further upkeeping of the feature, we have higher profile features to do instead :)

closed 2024-04-20 by danopernis