prusa3d/Prusa-Firmware-Buddy #3922: BUG: XL Printer goes offline when extruder heat command is send

ingbrzy, 2024-04-17:
Printer: Prusa XL 5 heads
Firmware: 5.1.2

In the PRUSA Connect nozzle temperature is shown as 6°C when all heads are parked
[image]

When all heads are parked and I send command via PrusaConnect Control to preheat NOZZLE (any temp), printer goes instantly OFFLINE and error is shown on the screen

[image]

eltron247, 2024-04-19:
I think I've found another place where this issue pops up directly on the printer.

I'm using 6.0.0 RC3 on a 2 tool XL.

When heating a tool to 240, 250, or 230, (I'm sure it works at any temp above idle) via either, first, manually setting the tool temp or, preheating the system, or directly telling it to load a filament and then choosing a temperature, or going to > Move Axis > then setting the temp:

The printer will heat the tool correctly but after waiting, what appears to be the dwell time, the printer times out, crashes / resets, and then warns of a bad heater or temp sensor wire issue. 

It seems to escape to the the watchdog somehow and never move past the safety checks in the preheat cycle. 

The system will run cycles normally and not experience this issue but when attempting to manually interact with the tools nozzle temp it leads, invariably, to an error.

This was tested on tool 1 if that makes a difference.

danopernis, 2024-09-06:
@ingbrzy @eltron247 can you test on 6.1.3 and confirm if this is still an issue?

DoubleStrike, 2024-10-06:
Still an issue.  The temp starts going up and then a PrusaConnect error/Printer error occurs requiring a reset:
[image]

suprchunk, 2024-11-09:
I receive the same, only through preheating via PrusaConnect. Every time I preheat via PrusaConnect I get this error. Making control via PrusaConnect a problem.

closed 2025-01-20 by github-actions[bot]