prusa3d/Prusa-Firmware-Buddy #3923: [BUG] MK4 6.0.0 Nozzle cleaning always fails

bgyarfas, 2024-04-17:
**Printer type** - MK4

**Printer firmware version** - 6.0.0

**Original or Custom firmware** - Original

**Optional upgrades** - Prusa v6 Adapter w/ 0.6 Diamond Back Nozzle

**USB drive or USB/Octoprint** - USB drive

**Describe the bug**
  Printer attempts nozzle clean with 10+ taps, fails every time. Restarting printer did not help. 

**How to reproduce**
  Firmware updated to 6.0.0. Attempting to print with Prusament PLA on Prusa Satin Sheet. File was sliced in Prusaslicer 2.7.4 with stock printer settings. Official v6 adapter w/ 0.6 diamond back nozzle. 

  Downgrading to 5.1.3 resolves the issue.

**Expected behavior**
  Nozzle cleaning to not fail :)

**G-code**
  BGCode is attached, with added `.txt` suffix so Github would attached it. 

[Test Fit Eye Holder_0.6n_0.25mm_PLA_MK4IS_9m.bgcode.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15013166/Test.Fit.Eye.Holder_0.6n_0.25mm_PLA_MK4IS_9m.bgcode.txt)

mikelane, 2024-04-17:
I was having this problem on the previous release. Is this still happening with 6.0.0-rc3?

mikelane, 2024-04-18:
Like I said, with RC2, I had the same problem with the failure to clean. But I just installed RC3 and I'm running a test print right now which seems to be working fine.

alphen87, 2024-04-18:
I have have the issue as well on the XL, first I thought it was a calibration issue, so I runned the nozzle offset. It does a lot of taps on the pin in z direction, but eventually completed. It did not seem to help (and also didn't matter which tool I selected) as it was still failing to get past the nozzle cleaning phase. 

After rebooting the machine, and running the calibration again, it started working again.

bgyarfas, 2024-04-18:
@mikelane This is happening for me on the official 6.0.0 release (not an RC).

bgyarfas, 2024-04-18:
If there are any logs/tests someone at Prusa would like me to run, I'd be happy to help.

danopernis, 2024-04-18:
@bgyarfas @mikelane @alphen87 Thanks for the bug report. I see several similar issues open already, so let's keep all the relevant info in one issue, which will be https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3818 as it was the first as far as I can tell.

closed 2024-04-18 by danopernis

TomaszVirco, 2025-06-04:
same here.. after upgrade this is persistant issue for each reprint.. clicking retry continue print with no issue