prusa3d/Prusa-Firmware-Buddy #3924: [BUG] XL Nozzle Cleaning Error with 6.0.0 RC3

BillC58, 2024-04-17:
The nozzle cleaning problem reported [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3818) is unresolved.

The biggest problem with a reset to continue printing is that the heat absorption routine is implemented again with a significant delay in printing.

danopernis, 2024-04-17:
Thank you for your report. I am closing this in order to not clutter the already-quite-long list of issues. Please note that creating duplicate issues does not help solving the original issue.

closed 2024-04-17 by danopernis

BillC58, 2024-04-17:
Sorry if I did the wrong thing. From the release notes, RC3 had bug fixes so I assumed you thought this one was fixed and I wanted to make sure that you knew it was still there.

 

The bug is also not showing as being assigned, or any action. I wanted to make sure it didn’t fall through the cracks.  I see others are reporting the issue too.

 

Thanks,

Bill