prusa3d/Prusa-Firmware-Buddy #3926: [BUG] MK4 6.0.0 Firmware update leaves USB drive empty and unreadable

ChiefPoints, 2024-04-18:
**Printer type** - MK4

**Printer firmware version** - 6.0.0

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint**
USB Drive

**Describe the bug**
After updating the MK4 firmware to 6.0.0 the USB drive is left empty and unreadable by computer, which caused the loss of ALL files.

**How to reproduce**
 Upgrade the firmware to 6.0.0

**Expected behavior**
I expected the firmware update to leave my files on the USB drive intact

dhyananbho, 2024-04-18:
When I insert my MK4 stick into the MacBook Pro, the Mac recognizes the stick, but no files are visible in the Finder. That should not be the case. However, when the stick is in the MK4, all files are present and also visible via Prusa Connect. This was already the case with FW 5.1.3 and is still the case after the update to FW 6.0. In other words, I carried out the update via Prusa Connect.

ChiefPoints, 2024-04-18:
> When I insert my MK4 stick into the MacBook Pro, the Mac recognizes the stick, but no files are visible in the Finder. That should not be the case. However, when the stick is in the MK4, all files [...]

Everything was fine and visible for me prior to the 6.0.0 update, no issues reading the drive on MacOS.  The update completely deleted the volume making the USB Drive like it was brand new, so nothing visible on the printer or the computer, so I had to recreate and format the volume.

smurfix, 2024-04-19:
This might be a firmware problem with partitioned sticks. I had the same difficulties with both the MK4 and the XL. They vanished after reformatting the sticks so that they don't have a partition table.

Under Linux you do that with `sudo mkfs.vfat -I -n MK4 /dev/sdX`. (Check `/proc/partitions` for the correct X. Don't use `/dev/sdX1`, that's the whole point of this exercise.)

m-cas, 2024-04-19:
it happened to me twice with previous updates plus it was REALLY SLOW to write to. I put in a new usb stick (verbatim, but I guess any decent brand will do) and never had a problem since. Given the price of usb stick nowadays it is not worth the trouble keeping the prusa one. Additional comment: given the same argument, I am surprised by the abysmally low quality of the supplied stick.

closed 2024-07-10 by github-actions[bot]