prusa3d/Prusa-Firmware-Buddy #3927: [BUG] Fw 6.0.0 PrusaMini unnecessary initial calibration and z axis self test fail

huskyte, 2024-04-18:
**Printer type** - [MINI]

**Printer firmware version** - [6.0.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Revo Hotend V2]

**Describe the bug**
  After the firmware update some self tests aka calibration seemed to be reset. One of them being the z axis self test. This failed although passed successfully before and any motion is smooth. Additionally, the missing calibration does not let the user print anymore without manually rejecting the warning on the printer.

**How to reproduce**
  Rather obvious

**Expected behavior**
  1. Self tests, once passed, stay passed and maybe get a one time warning to be repeated if changes were made by Prusa
  2. The missing calibration warning can be turned off somewhere in the (advanced) menu in order to let the printer function properly
  3. Failing a selftest either dumps the raw data onto USB in human readable format or displays comprehensive information 

**G-code**
 none

**Crash dump file**
  I would love to provide one to see the actual error, but this seems not to be possible

ClaGre, 2024-04-20:
> **Printer type** - [MINI]
>
> **Printer firmware version** - [6.0.0]
>
> **Original or Custom firmware** - [Original]
>
> **Optional upgrades** - [Revo Hotend V2]
>
> **Describe the bug** After the firmware [...]

Same issue here, same configuration: Revo Mini and Vertex Extruder

[image]

huskyte, 2024-04-21:
Little more info. I couldn't manually measure the z hight, because it doesn't let you control the z axis to the end. But I stuck a cable tie on top of the z axis carriage, intently blocking off .5 mm and the calibration passed. Usually it stalls with z height 0.39 mm, but this time it passed with .38.

ClaGre, 2024-04-21:
> Little more info. I couldn't manually measure the z hight, because it doesn't let you control the z axis to the end. But I stuck a cable tie on top of the z axis carriage, intently blocking off .5 [...]

To misure de Z axis you can use a simple meter 🤭 but happy you've solved 👍🏻

YoannH8437, 2024-04-27:
Same bug for me.
Appendix still on, but Sherpa mini direct drive installed.

ClaGre, 2024-04-28:
> Same bug for me.
> 
> Appendix still on, but Sherpa mini direct drive installed.

What is your gear ratio (Sherpa mini) you’ve installed?
What is your steps/mm had you set into firmware?

YoannH8437, 2024-04-28:
> > Same bug for me.
> > Appendix still on, but Sherpa mini direct drive installed.
> 
> Please two questions not connected to your problem: What is your gear ratio (Sherpa mini) you’ve installed? What is your steps/mm had you set into firmware?

Hello.
It can be linked: I changed the top holder of z column to avoid collision with the extruder motor.

It is a 6.25:1
I entered 880steps/mm. [Repeated in the start g-code. M92 E880]
Plus current in start g-code set to 700mA [M906 T0 E700]

Tarudro, 2024-05-04:
I had the same issue. Was quite surprised my prusa didn't pass the self check, because I was printing very successfully for a long time.

I just want to quickly add also how I solved it for now, such that others can be quicker as well:

Go to your Settings --> Hardware and long-press the menu. Then you will have more options available and also a setting to change the Z-Axis length. Default value (for me) was 185. I changed it to 190, which made the self test pass.

YoannH8437, 2024-05-04:
> I had the same issue. Was quite surprised my prusa didn't pass the self check, because I was printing very successfully for a long time.
>
> I just want to quickly add also how I solved it for now, [...]

Reading a bit previous answer and finding clipped in spacers to put on z rods I guessed it was linked to this parameter.
If I stop z at 180mm I have 12mm left. So 191 or 192 should work for me. I will try this before going prusa mini bear or ratrig alike [30×60 extrusion, metal piece to keep everything square, linear rails&co] I am a bit bored of the flabby x axis and skew [I fixed it several time but petg at 37-40deg of the enclosure flow -"fluage sous contrainte" in french- a bit, so to redo periodically]. Especially with the deactivated skew compensation at firmware level.

sleeperninja, 2024-05-06:
Z axis failure here, as well. I have the Bondtech extruder.

closed 2024-10-21 by github-actions[bot]