prusa3d/Prusa-Firmware-Buddy #3930: [BUG] Stuck filament detection is too sensitive

smurfix, 2024-04-19:
**Printer type** - XL

**Printer firmware version** - 6.0.0 RC3

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint** USB

**Describe the bug**
In some situations the filament stuck detection assumes stuck filament where flow is unimpeded.

**How to reproduce**
The attached gcode has a very long zigzag pattern in layers 316 and 317. The printer errors out multiple times when it prints it, despite filament flowing normally.

**Expected behavior**
The model prints without issue.

When the sensor detects stuckage, offer the user an option to extrude some filament to check whether the blockage is real or just an over-sensitive sensor. If the latter, continue printing with reduced sensitivity, or with the sensor turned off entirely.

**G-code**
[spool.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15033580/spool.zip)

DRracer, 2024-04-25:
Filament stuck detection tries to detect skipping of E-motor's steps. If individual steps are skipped (for any reason including high flow rate and/or filament too cold), you might not notice if not looking at the drive gear directly (it's hidden inside the extruder obviously). But the algorithm triggers trying to save the print.

Setting the trigger threshold would be possible, but without real-time data it is very hard to dial better than we did.

We'll try printing the attached G-code to see what forces are we getting at the range of layers you specified.

For now, I'd advise increasing the nozzle temperature and/or lowering the print speed just a bit.

nikoladom91, 2024-05-09:
I have a similar issue on the MK4. I would not say that it is allways wrong in detecting a clog as i got a failed printe when disabeling the sensor. But, whenever it occures, it is enough to unload and reload the fillament and the purge usualy solves it.

Is it possible to have the printer automatically try purging fillament to the side to get rid of the clog? I feel that would have continued a lot of my paused prints.

bablokb, 2025-09-19:
> Filament stuck detection tries to detect skipping of E-motor's steps.

Can you explain why it detects stuck filament always at exactly the same position in the print? I just can't believe it is a problem of sensitivity. Or maybe the skipping of E-motor's step is deterministic.

I have a very simple model, it "detects" stuck filmament always at the same position. After unload/load it continues and it detects the stuck filament shortly later again at a fixed position. And if you read your knowlegebase you will find reports of people reporting the same exact thing. So it cannot just be the sensitivity, this does not explain the deterministic behavior.