prusa3d/Prusa-Firmware-Buddy #3934: [BUG] XL 6.0.0-RC3 sometimes restarts probing in the middle of the print

muesli, 2024-04-20:
**Printer type** - XL 5TH

**Printer firmware version** - 6.0.0-RC3

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
Printed via Prusa Link.

**Describe the bug**
While in the middle of printing the first layer (snug support structure), the printer suddenly re-started Z-homing, nozzle cleaning and mesh probing. At this point I aborted the print, as I was worried it would restart the entire print and crash into the existing structure on the bed.

**How to reproduce**
I'm not aware how to reproduce this, I've tried reslicing and restarting the same model and it seems to print fine at this point.

**Expected behavior**
Don't restart probing when in the middle of a print.

DRracer, 2024-04-25:
Please double check the G-code file on your USB flash drive. That's usually the culprit. The file is either incomplete, contains cross linked blocks or some other inconsistency.

muesli, 2024-04-25:
Is this a common problem with the "official" Prusa-labeled USB sticks that came with this printer?

murk-sy, 2024-04-29:
This happened a few times for me on the Mini but it's been a long while. Happened mostly on earlier firmwares I *believe*.
There should some sort of a check that prevents gcode that shouldn't be ran from running (auto bed leveling being one of them).

Related enhancement: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/830

muesli, 2024-05-05:
Just happened to me again with 6.0.0. The G-Code on the stick is fine, I've tried to simply reprint the very same file and it succeeded the second time. Something triggers the machine to restart the print entirely. This is absolutely fatal, as the machine will crash into the existing object.

DRracer, 2024-05-06:
Please provide your G-code (if you can) - copy it from your USB drive (not from your PC).
We haven't seen a similar case with the latest FW

muesli, 2024-05-13:
> Please provide your G-code (if you can) - copy it from your USB drive (not from your PC). We haven't seen a similar case with the latest FW

Here's the (b)gcode that caused the issue, generated by PrusaSlicer 2.7.4:

[main_body_0.4n_0.2mm_PLA_XLIS_2h47m.bgcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15297674/main_body_0.4n_0.2mm_PLA_XLIS_2h47m.bgcode.zip)

muesli, 2024-06-12:
In the meantime the official Prusa USB Stick that I used for these prints stopped working entirely, so I assume it has always been the culprit of this issue. Closing until I'm able to reproduce this with a new stick.

closed 2024-06-12 by muesli