prusa3d/Prusa-Firmware-Buddy #3935: [BUG] <Enter comprehensive title>

epetit91, 2024-04-20:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI, MK4, XL]

**Printer firmware version** - [e.g. 4.0.5, ...]

**Original or Custom firmware** - [Original/Custom]

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please send the crash dump file to Prusa by emailing it to reports@prusa3d.com. Sharing this file is important and helps us investigate the bug.

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

**Video**
  Please attach a video. It usually helps to solve the problem.

3d-gussner, 2024-04-22:
@epetit91 We have on the MK3S this translation `Déchargement du filament réussi?`
https://github.com/prusa3d/Prusa-Firmware/blob/MK3/lang/po/Firmware_fr.po#L2320-L2321

epetit91, 2024-04-22:
Yes, this Is correct too.Envoyé de mon iPhone

https://github.com/prusa3d/Prusa-Firmware/blob/MK3/lang/po/Firmware_fr.po#L2320-L2321

closed 2025-09-25 by 3d-gussner