prusa3d/Prusa-Firmware-Buddy #3936: [BUG] Material in Footer of MK4 not updated during multi-material print with MMU3

unclej84, 2024-04-20:
**Printer type** - [MK4]

**Printer firmware version** - [6.0.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [MMU3]

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
During a multi-material print with the MMU3 the material in the footer does not get updated according to currently used material although it was correctly defined by advanced loading

**How to reproduce**
Define multi-material print for MK4 with MMU3 using PETG in slot 1 and PLA in slot 2 - footer always stays at the same/first material

**Expected behavior**
[top-bolt-x-2_0.4n_0.2mm_PETG,PLA_MK4ISMMU3_1h16m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15050104/top-bolt-x-2_0.4n_0.2mm_PETG.PLA_MK4ISMMU3_1h16m.zip)

Display currently used material - when slot 1 is in use, display "PETG" - when slot 2 is in use, display "PLA"

**G-code**
[top-bolt-x-2_0.4n_0.2mm_PETG,PLA_MK4ISMMU3_1h16m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15050104/top-bolt-x-2_0.4n_0.2mm_PETG.PLA_MK4ISMMU3_1h16m.zip) - contains 3mf project and sliced bgcode file

VladaDHS, 2024-05-04:
Have same issue. all Fialments are PLA but down is shown PETG.

seifer69er, 2024-06-20:
After some more testing i think I know what the issue is,
When loading each filament into the MMU3, there is no filament type option on the MK4, so it relies on what it thinks is in that slot,
For me, slot 1 was PLA, 2 was  ---, 3 ----, 4 ---- and 5----
so when i use prusaslicer  ive told the slicer what tool is which filament.
the issue is the filament type not displaying correctly is that the Mk4 relies on EEPROM, for this data and not prusa connect

I did manage to select the correct filament type on the device, and it managed to load just fine, but otherwise its a fun bug to debug :)

unclej84, 2024-06-20:
The filament type selection is available with advanced loading via the menu - I used this and specified the material correctly, so the MMU/MK4 knew the type but still showed the wrong type

unclej84, 2024-08-20:
Is there any update from Prusa side?

danopernis, 2024-08-20:
Yes, this is confirmed, but it is not very high on the priority list.

CZDanol, 2024-08-30:
Internal ticket: BFW-6014