prusa3d/Prusa-Firmware-Buddy #3939: [BUG] Systematic Z levelling problem

smurfix, 2024-04-21:
**Printer type** - XL, multi-head

**Printer firmware version** - 6.0.0 RC1, but also happens with 5.x

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint** - both

**Describe the bug**
My printer exhibits consistent levelling errors (Z offset too small) on the right side of the print area, but *only* if the print extends to the left side.

**How to reproduce**
See the attached 3MF file.

[image]

When I only print fields 3+6+9+12+14, fields 6 and 9 are basically perfect.

When I print all fields, or only 7+8+9+13+14, fields 6 and 9 are overextruded, obviously due to too little Z offset as determined by auto levelling.

This is 100% reproducible on my printer and does not depend on selected extruder, filament type, dirt on the bed or the nozzle (which IMHO could only result in underextrusion anyway), the relative position of right vs. left Z axis, or any other influence I could discover.

An overextruded right edge might not look like much, but here's a screenshot from a video of a larger print:

[image]

Observe the perfectly-extruded left side vs. the obvious and pervasive overextrusion on the right side.

**Expected behavior**
Fields 6 and 9 are always perfectly levelled. My large print doesn't show overextrusion.

[First Layer.3mf.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/15051878/First.Layer.3mf.zip)

smurfix, 2024-04-21:
Another test: after printing 1+4+6+9, field 6 is obviously over- and field 1 equally obviously underextruded.

[image]

m-kozlowski, 2024-04-21:
This looks a lot like a miscalculated compensation problem due to PINDA sampling too close to the right edge of the bed mentioned in issue #3851

smurfix, 2024-04-21:
Well except for (a) the XL doesn't have a PINDA and (b) the problem in 3851 is hardware and doesn't fix itself when you limit your print to the area affected by the problem.

I agree that the most likely cause is some sort of miscalculation, however I'm certainly not the only person who covers most of the build plate with a first layer, thus there's bound to be *something* about my printer that's different enough for the calculations to throw themselves off.

I have no idea what, though, given that prints are perfectly fine when you confine them to the right *or* the left side of the bed.

m-kozlowski, 2024-04-21:
> the XL doesn't have a PINDA

my bad. I havent noticed your printer model

> the problem in 3851 is hardware and doesn't fix itself when you limit your print to the area affected by the problem.

Oh, but it does:)

flymac, 2024-06-19:
Hello
Same problem on MK3.5 after upgrade. Spent countless hours with the support to no avail. The nozzle is too high on the left side of the plate. Trying with bed corrections but it's very inconsistent. Firmware 6.0.2.