prusa3d/Prusa-Firmware-Buddy #3943: [ENHANCEMENT]<Prusa XL MultiTool - Cancel Object to stop unneeded toolhead changes and purging>

brianirwin, 2024-04-23:
**Printer type** Prusa XL 5-Tool, Prusa Slicer 2.7.4+win64, Firmware 5.1.2.

**Is your enhancement related to a problem? Please describe.**
If you cancel all objects in a print that require a tool-change, on subsequent layers the printer still switches tool, primes the no longer needed tool(s) on the wipe tower and then switch back to the tool that is required. Wipes on the tower, and resumes the print.

**Describe the solution you'd like**
Once all objects that are required for a tool are cancelled, the printer should not grab / prime that tool again. If only a single tool is now required, no further use of the purgeblock/wipetower should be done.

murk-sy, 2024-05-03:
**Related issues:**
- https://github.com/prusa3d/PrusaSlicer/issues/11615
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3807
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3583

Would likely be a bit trickier to implement with either a list of future tools required listed around every toolchange or at least layer change. Lookahead through gcode is likely a bit much even for the xl soc, but it only has to run once every object cancellation.

brianirwin, 2024-05-08:
@murk-sy I do not know enough about how slicer/cancel object works to have a valid opinion. But your second suggestion seems the more correct. 

My simple vision of how you might do something like this, is write in slicer a cheat-sheet for each object and the final layer it uses a tool.

 Object 1, Tool 1 to layer 100, Tool 2 to layer 50, Tool 3 to layer 200.
 Object 2, Tool 1 to layer 500
 Object 3, Tool 2 to layer 60
 Purge Tower, Tool 1 to 500, Tool 2 to 60

As a part of cancel-object logic it compares current layer height to the lookup table. if you are above the last time that tool gets used that tool never gets selected again. 

Again apologies if I am overstepping, I know the comment was not for me, but it felt like I could offer some more context :)

closed 2024-07-15 by github-actions[bot]